respeaker/get_started_with_respeaker #1: Android

scargill, 2016-08-23:
DidI miss something - the examples of firing music at the device are for Apple. What about Android - do you have examples.

xiongyihui, 2016-08-24:
With Andriod, DLNA is supported, we will add a guide later. Besides, AirPlay is supported by android apps such as AllConnect

scargill, 2016-08-24:
And the missing .py file?

scargill, 2016-08-24:
This needs to be more concise….

 

I have the reSpeaker running – the AllConnect is getting awful reviews so I do not want to put this on my phone. I added Airiwire to the phone and it is seeing my Plex Media server, but not the reSpeaker unit that is sitting connected to my network.  You should not expect people to have to guess how to use this.

 

So – how is this connected to Android by DLNA apps such as Airwire  - is the missing .py file available now?

Regards

 

Pete.

scargill, 2016-08-25:
Need example of Android use – you are asking people to do much R&D on this… and still missing one of the files in your guide.

xiongyihui, 2016-12-07:
With the latest firmware, DLNA is enabled by default, connect to the respeake Access Point and use DLNA with  AllConnect or BubbleUPnP

closed 2016-12-07 by xiongyihui