respeaker/get_started_with_respeaker #3: pyaudio install fail.

eungyoda, 2016-09-21:
I have a problem during the respeaker test. 
I work after a factory reset.

I currently go through the steps to install pyaudio. like this -> [Python packages on Respeaker](https://github.com/respeaker/get_started_with_respeaker/wiki/Python-packages-on-Respeaker)
 
However, the following error occurs.
 
> building '_portaudio' extension
> creating build/tem.linux-mips-2.7
>  
> creating build/temp.linux-mips-2.7/src
 
> ccache_cc -fno-trict-aliasing -Os -pipe -mno-branch-likely -mips32r2 -mtne=24kec -mdsp -fn-caller-saves -fhonour-copts -Wno-error=unused-but-set-vaiable -msoft-float -DNDEBUG -fno-inline -DNDEBUG -Os -pipe -mno-branch-like -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fhonour-copts -Wno-error=used-but-set-variable -msoft-float -fPIC -I/usr/include/python2.7 -c src/_poaudiomodule.c -o build/temp.linux-mips-2.7/src/_portaudiomodule.o
> unableo execute 'ccache_cc': No such file or directory 
> error: command 'ccache_' failed with exit status 1
 
It does not seem to "ccache_cc"  on the current respeaker linux. 
Please tell me the answer for this problem.

xiongyihui, 2016-09-21:
PyAudio requires cross compilation. We write a [Makefile](https://github.com/respeaker/respeaker-feed/tree/master/python-pyaudio) to build it. It's included in [this firmware](https://drive.google.com/open?id=0BwBIy7Eq4NucT1FrRGV5Y0NfNVU) which can be used directly

eungyoda, 2016-09-23:
hi @xiongyihui  I tried a firmware upgrade. but this Image files are incorrect.
Please check again.

xiongyihui, 2016-09-23:
How did you try? The image is compressed, extract it first, then copy it to openwrt's `/tmp` and run `sysupgrade`

closed 2016-09-27 by xiongyihui