respeaker/get_started_with_respeaker #9: Inconsistent data reading via HID interface from Raspberry Pi

SebMiller, 2016-12-09:
### Setup:
- Raspberry Pi 3 Model B V1.2:
```
$ uname -a
Linux raspberrypi 4.4.34-v7+ #930 SMP Wed Nov 23 15:20:41 GMT 2016 armv7l GNU/Linux
```
- pyusb:
```
$ pip show pyusb
---
Name: pyusb
Version: 1.0.0
Location: /usr/local/lib/python2.7/dist-packages
Requires: 
```
- respeaker:
```
$ pip show respeaker
---
Name: respeaker
Version: 0.4.2
Location: /usr/local/lib/python2.7/dist-packages
Requires: 
```
- ReSpeaker Microphone Array v1.0 08/29/2016
- ReSpeaker connected to Raspberry Pi MicroUSB cable

### Description of the problem:
Problems appear when reading register values using the mic_array.py sample program.
```
$ sudo python mic_array.py w 0x0000 0x05050501
$ sudo python mic_array.py r 0x0000
[158, 55, 165, 16]
```
We would expect to receive `[5, 5, 5, 1]` in this case.
Also, reading any other address always returns the same values:
```
$ sudo python mic_array.py r 0x0003
[158, 55, 165, 16]
$ sudo python mic_array.py r 0x0004
[158, 55, 165, 16]
```
 
### Also to be noted:
- After plugging the ReSpeaker Mic Array into the Raspberry Pi, the two first reads are always different, but from the third read, the same values are always returned:
```
$ sudo python mic_array.py r 0x0008
[0, 0, 0, 0]
$ sudo python mic_array.py w 0x0008 0x00010203
$ sudo python mic_array.py r 0x0008
[1, 0, 0, 0]
$ sudo python mic_array.py w 0x0008 0x000a0b0c
$ sudo python mic_array.py r 0x0008
[158, 55, 165, 16]
$ sudo python mic_array.py w 0x0008 0x00a1b2c3
$ sudo python mic_array.py r 0x0008
[158, 55, 165, 16]
```
- After modifying the mic_array.py sample program, the voice angle and VAD state where correctly read from the REPORT_AUTO message, but not when reading directly at the address of the value (0x44 for voice angle, 0x40 for VAD state).

Thank you in advance for your investigation into this. Please feel free to ask for additional information if needed.
Best regards,
Sebastian Miller
YUKAI Engineering

xiongyihui, 2016-12-09:
I wrote the script on windows. It seems the script doesn't work on linux. For Linux, please try the javascript library - [`node-hid`](https://github.com/node-hid/node-hid) at the moment.

seantrue, 2017-01-16:
Since many of us do not run anything on Windows, working Linux code would be useful.

KillingJacky, 2017-10-16:
Sorry we haven't enough resource at that time, hope you have come over this problem later. FYI, we have a section about "HID for controlling ReSpeaker Mic Array" in the [wiki page](http://wiki.seeed.cc/ReSpeaker_Mic_Array/#hid-for-controlling-respeaker-mic-array).

closed 2017-11-27 by KillingJacky