respeaker/get_started_with_respeaker #10: Authorisation error

lawrie, 2017-01-06:
My Respeaker worked for a while but is now giving 

Access to 192.168.100.1 was denied

You don't have authorisation to view this page.
HTTP ERROR 403

I can log into it by a serial console, and it is getting an IP address from my router.

Speech recognition did not work for me as the Bing Speech recognition API seems to be down at the moment.

JohnOH, 2017-01-06:
I'm pleased you got it working for a while. I haven't managed to get anything to work yet.

lawrie, 2017-01-06:
Alexa works for me on it now as well. But the audio is a bit clipped and it is slower than the real thing.

It took a few goes to get the device working. The login.html page that allows you to connect to your Wifi router wasn't coming up at first - it went straight into the JuciBox dashboard. But I shut it down and restarted it a couple of times and it eventually worked.

The 192.168.100.1/home.html page works OK if I connect my PC to the Respeaker access point but no longer works when I connect to my Internet router. (As I say, it did for a while). I suspect that the javascript is failing somewhere.

The firmware didn't need updating - it said it was the latest version.

I get the authorisation error when I access the page fro my PC, but I get a blank page when I use  Chrome on Ubuntu.

https://speech.platform.bing.com/ is currently giving::

>  <h2>Our services aren't available right now</h2><p>We're working to restore all services as soon as possible. Please check back soon.</p>Ref A: D827E7EDE37B432FAEF7798EB3181FAF Ref B: LTSEDGE0122 Ref C: Fri Jan  6 13:35:47 2017 PST

which is why I don't think the python speech recognition example is currently working.

The python example gives:

> INFO:requests.packages.urllib3.connectionpool:Starting new HTTPS connection (1): oxford-speech.cloudapp.net
> DEBUG:requests.packages.urllib3.connectionpool:"POST /token/issueToken HTTP/1.1" 401 143
> 111000000000000000000000000000000000000000000000000000000recognition connection failed
> 

when I say, "Respeaker play music".

I got a few alsa channel errors when I first tried the python example, and had to restart the device to make them go away.

JohnOH, 2017-01-06:
I also had the error you describe. Ultimately I couldn't get past a message about there not being a default input device. I have assumed that the headphone jack can be used to hear the output as I haven't soldered in a speaker. I have tried Bing, standalone an Alexa - I had to put it to one side today as it was getting me quite angry.

lawrie, 2017-01-07:
I used a rechargeable battery-powered speaker via headphone jack. (The type of speaker designed for an ipod or other mp3 player and which I use with Raspberry Pis).

I am now unsure if I ever had it working via 192.168.100.1 while connected to my router. I might just have seen cached pages. The instructions are unclear about whether you are supposed to leave it connected to the Respeaker access point or connect back to your internet Wifi router. Connecting to the Respeaker access point whenever you use the web interface is not  practical. It works if I add a route to my PC for 192.168.100.1. Most such devices use mDNS and .local addresses to make it easy to connect to.  Not sure if a .local address is available with this version.  I have also been setting up an  Onion Omega 2 device today which also runs OpenWRT. That uses .local addresses. 

I don't think I saw a message about no default input device.

JohnOH, 2017-01-07:
When the WAN wifi connects to your main router the ReSpeaker functions as a router between 192.168.100.x segment and your main router LAN side segment from there it can get to the internet. 

Are you using the internal microphone or the add on array microphone. I was wondering if it defaults to the array microphone and then since I don't have one it fails to find the internal one.

lawrie, 2017-01-07:
I don't have the microphone array.

I know the device acts as a router. My issue was getting to 192.168.100.1 from a PC or other device when the PC  is connected to my main router.  I can use the IP address allocated to the Respeaker by my main router DHCP, but I need to find what IP address has been allocated. I could use a .local address if it was supported. I can get to 192.168.100.1 if I add a route to by PC, making the Respeaker allocated IP address a gateway to the 192.168.100.x subnetwork, but the instructions on Github made it look like this would work by default. I think they need updating.

JohnOH, 2017-01-07:
There is a firewall between the WAN side and LAN side of the ReSpeaker it appears to be configurable via 192.168.100.1 (connected to ReSpeaker wifi). Once adjusted it should be possible to get into the ReSpeaker from is WAN side. Could make it a static address rather than sticking with the address assigned by your router - some segment address obviously.

lawrie, 2017-01-08:
I noticed that the avahi service is running on the respeaker, and that I can connect to the console by http://respeaker.local.

AeroFont, 2017-03-31:
I've had my ReSpeaker since the week started and its been a roller coaster ride of emotions. I was quite excited, and have been trying my hardest, but there is always something breaking. Either its the Interface, the runtime behind it all, the WiFi, the SD Partition, the bootloader or python. The only thing Ive got to run this week was the offline Speech recognition, but that stopped working after a reboot.

closed 2018-08-01 by lawrie