respeaker/get_started_with_respeaker #11: Invalid number of channels

lawrie, 2017-01-07:
I often get this error when attempting to do speech recognition, e.g when using alexa:

> root@ReSpeaker:~# alexa
> Expression 'parameters->channelCount <= maxChans' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1513
> Expression 'ValidateParameters( inputParameters, hostApi, StreamDirection_In )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2813
> Traceback (most recent call last):
>   File "/usr/bin/alexa", line 288, in <module>
>     main()
>   File "/usr/bin/alexa", line 262, in main
>     mic = Microphone(quit_event=quit_event)
>   File "/usr/lib/python2.7/site-packages/respeaker/microphone.py", line 89, in __init__
>     input_device_index=self.device_index,
>   File "/usr/lib/python2.7/site-packages/pyaudio.py", line 750, in open
>     stream = Stream(self, *args, **kwargs)
>   File "/usr/lib/python2.7/site-packages/pyaudio.py", line 441, in __init__
>     self._stream = pa.open(**arguments)
> IOError: [Errno -9998] Invalid number of channels

It usually goes away if I reboot the device.

sash13, 2017-01-07:
Something similar problem here:
http://stackoverflow.com/questions/31073667/getting-ioerror-errno-invalid-number-of-channels-9998-when-using-mic-with-py

Take picture of command 
$  alsamixer

xiongyihui, 2017-01-08:
It should be that mopidy-hallo is running and using the audio input device.
To use alexa, stop mopidy by running `/etc/init.d/mopidy stop && /etc/init.d/mopidy disable`

lawrie, 2017-01-08:
I will try that. I got most success by doing /etc/init.d/alexa stop, then alexa, then waiting a long while. When alexa does work the start of the audio is cut off and things like playing radio stations don't work. That always seems to get a synchronization error. 

But the bing speech API doesn't seem to be working at all at the moment.

lawrie, 2017-01-08:
This what is produced when I say "Alexa, play bbc radio 4". It replies, "BBC Radio 4 from Tune In", and then stops:

> DEBUG:root:Start sending speech to Alexa Voice Service
> INFO:mc:Stop listening
> DEBUG:rot:Finished sending speech to Alexa Voice Service
> MPEGAudio Decoder 0.15.2 (beta) - Copyright (C) 2000-2004 Robert Leslie et al.
> error: frame 0: lost synchronization
> Playing WAVE'stdin' : Signed 16 bit Li[  273.420000] playback free_dma_buffer
> ttle Endian, Rat[  273.430000] ptri2s_config->mmap_index:8
> e 24000 z, Mono
> error: frame 82: lost synchronization
> 82 fames decoded (0:0:01.9), -1.6 dB peak amplitude, 0 clipped samples
> [  275.650000] ummap MMAP[0]=0x86390000
> [  275.660000] playback free_dma_buffer
> [  275.670000] ptri2s_config->mmap_index:
> INFO:mic:Start detecting
> INFO:mic:Too many delays, 8 in queue
> INFO:mic:Too many delays, 7 in queue
>

xiongyihui, 2017-01-09:
It should be a bug. Thanks for the report.

Bing Speech API is fixed now.

lawrie, 2017-01-09:
I still get this when I use the python example with Bing Speech API:

> INFO:mic:etected respeaker
> 000000000000000000000000000000000000000000000000000Wake up
> INFO:mic:Start listening
> got data
> DEBUG:requests.packages.urllib3.connectionpool:"POST /token/issueToken HTTP/1.1" 401 143
> 0000000000000000recognition connection failed

closed 2017-01-16 by xiongyihui