respeaker/get_started_with_respeaker #12: Failed to connect

svendar, 2017-01-08:
Hello

I am having problems to connect the respeaker to the wireless network.
I tried it via the Wifi settings page. It can see my wireless network. When i enter the password and try to connect it says: Failed to connect.

Afterwards i tried via the command line command wictl like they mentioned:

`root@ReSpeaker:~#wictl

0,telenet-11C4E
1,TELENETHOMESPOT
2,WiFi-2.4-8220
3,TelenetWiFree
4,PROXIMUS_FON
5,telenet-CDF3AA2
6,telenet-117E3

Please choose your wifi:5
Please input the wifi password:-----------------

udhcpc(v1.23.2)started
Sendingdiscover...
Sendingdiscover...
Sendingdiscover...
Nolease,failing
ioctlerror
:Cannot assign requested address
ioctlerror
:Cannot assignrequested address
ioctlerror
:Cannot assign requested address
failed`

anyone an idea?

xiongyihui, 2017-01-08:
It seems the respeaker can't get an ip from your wireless router.

svendar, 2017-01-08:
this is what is found in my wireless config

root@ReSpeaker:~#cat/etc/config/wireless

config wifi-device'radio0'optiontype'ralink'
	option variant 'mt7628'
	option country 'TW'
	option hwmode '11g'
	option htmode 'HT40'
	option channel 'auto'
	option disabled '0'
config wifi-iface 'ap'
	option device 'radio0'
	option mode 'ap'
	option network 'lan'
	option ifname 'ra0'
	option encryption 'none'
	option ssid 'ReSpeaker1DD6F1'
	option seq '1'
config wifi-iface 'sta'
	option device 'radio0'
	option mode 'sta'
	option network 'wan'
	option ifname 'apcli0'
	option led 'mediatek:orange:wifi'
	option ssid 'UplinkAp'
	option key 'SecretKey'
	option encryption 'psk'
	option disabled '1'
	option ApCliEnable '1'
	option ApCliAuthMode 'WPA2PSK'
	option ApCliSsid 'telenet-CDF3AA2'
	option ApCliWPAPSK'-------------'

config wifi-status 'status'
	option wps '1'
	optionsched_status '0'
	option schedule '0'
	option wlan '1'

it seems like my connection is disabled (option disabled '1')

svendar, 2017-01-08:
i think there was a problem with my router
i did a reset of the router and now it works

closed 2017-01-08 by svendar