respeaker/get_started_with_respeaker #13: Bing Speech API not working

lawrie, 2017-01-09:
I started a new issue for this.

When I try the sample python program and say "Respeaker play music" I get:

> INFO:mic:etected respeaker
> 000000000000000000000000000000000000000000000000000Wake up
> INFO:mic:Start listening
> DEBUG:requests.packages.urllib3.connectionpool:"POST /token/issueToken HTTP/1.1" 401 143
> 0000000000000000recognition connection failed

I am using a key generated as explained in the instructions. There are two keys generated Key 1 and Key 2 as shown in the instructions. I am using Key 1. Is that correct?

It may be that the Bing Speech API is currently not available.

xiongyihui, 2017-01-09:
Are you using https://github.com/respeaker/microsoft_cognitive_services? It's a little old.
Please use https://github.com/respeaker/respeaker_python_library/blob/master/respeaker/bing_speech_api.py instead.

lawrie, 2017-01-09:
Yes I am using that cognitive services API.

After I do 

export BING_KEY="<my key>"

The bing_speech_api.py example works. But the example in the start-up instructions still doesn't.

lawrie, 2017-01-09:
When I change playmusic.py to use the version of BingSpeechAPI in my copy of bing_speech_api.py  it works. 

As I put the files in the same directory, I just have to remove "respeaker." from the import.

gavinc, 2017-01-26:
For lucky travellers:
The respeaker version available via pip is out of date. Clone the referenced git repo and do the "python setup.py install" of this up to date respeaker library, your respeaker python lib will be updated and the examples will once again work.

quezbeme, 2017-02-16:
> Are you using https://github.com/respeaker/microsoft_cognitive_services? It's a little old.
Please use https://github.com/respeaker/respeaker_python_library/blob/master/respeaker/bing_speech_api.py instead.

I'm getting the same issue here. I have the newer py but i still get the same 401 error. Am i missing something?

acidhax, 2017-04-25:
root@ReSpeaker:~/respeaker_python_library# python setup.py install
running install
Segmentation fault

xiongyihui, 2017-04-26:
@acidhax The library is installed by default. You don't have to re-install it.

closed 2017-10-16 by KillingJacky