respeaker/get_started_with_respeaker #14: Netgear R7000 SSID not seen my ReSpeaker

JohnOH, 2017-01-09:
Is there something specific about the ReSpeaker and its WAN connection to the Internet. I do not have problems using the wifi on my Netgear R7000 which supports 2.4G and 5G wifi. I would expect the ReSpeaker to be able to detect the SSID of the 2.4G wifi on the Netgear but it cannot. The ReSpeaker can see the wifi on another model of router. How can the incompatibility be resolved.

JohnOH, 2017-01-09:
Also noted that when connected to ReSpeaker wifi the SSID or my Netgear R7000 networks are no longer visible on my MacBook. I need to disconnect from ReSpeaker wifi before I am able to see and then connect to my more usual wifi connection.

xiongyihui, 2017-01-09:
Can you find out which Wi-Fi channel do the respeaker or the router use? You can use android apps to find these channels, such as https://play.google.com/store/apps/details?id=com.vrem.wifianalyzer&rdid=com.vrem.wifianalyzer

JohnOH, 2017-01-09:
ReSpeaker (192.168.1.100) is on channel 4.
ReSpeaker is using channel 6 on its WAN side.
My Netgear R7000 is operating on channel 12.

xiongyihui, 2017-01-16:
Channel 12 is not available at some countries and may not be supported by some devices.
It seems the ReSpeaker only support one channel. The channel of ReSpeaker LAN will stay the same with the channel of its WAN.
But I have no idea why the ReSpeaker changes the visible of the netgear router. It's weird!

closed 2017-10-16 by KillingJacky