respeaker/get_started_with_respeaker #16: Errors when opening microphone

lawrie, 2017-01-11:
When calling:

> mic = Microphone(quit_event=quit_event)

from python, I often get.

> Expression 'parameters->channelCount <= maxChans' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1513
> Expression 'ValidateParameters( inputParameters, hostApi, StreamDirection_In )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2813

I sometimes get a segmentation fault. Occasionally, it works.

It works OK if mopidy is not running, but as I am trying to control mopidy by voice command via an mpc client, this is not an option.

If there is a better place to raise these issues, please let me know.

lawrie, 2017-01-11:
The only way I got this to work is to make my program:

- stop mopidy 
- connect to the microphone
- start mopidy
- wait for it to start and then connect to it

xiongyihui, 2017-01-11:
There is a mopidy plugin `mopidy-hallo` which uses micrphone too. You can modify the mopidy-hallo to control the music player.

closed 2017-01-16 by xiongyihui