respeaker/get_started_with_respeaker #18: how to change keyword

xiaooquanwu, 2017-01-16:
how to change keyword ?

Nos77, 2017-02-14:
I'm interested too.

I have tried using this with an extended dictionary found here  https://github.com/respeaker/pocketsphinx-data
and this script : 
`respeaker_python_library/examples# python offline_voice_assistant.py`

But I got those errors and respeaker keep waking up even if i say nothing...

```
INFO:mic:Start detecting
INFO:mic:Too many delays, 5 in queue
INFO:mic:Too many delays, 5 in queue
INFO:mic:Too many delays, 6 in queue
INFO:mic:Too many delays, 7 in queue
INFO:mic:Too many delays, 7 in queue
INFO:mic:Too many delays, 8 in queue
INFO:mic:Too many delays, 9 in queue
INFO:mic:Too many delays, 9 in queue
INFO:mic:Too many delays, 10 in queue
```

svissarapu, 2017-04-03:
Yes, Can you please help with documentation for changing the wake up work?

KillingJacky, 2017-10-16:
Please refer to http://wiki.seeed.cc/ReSpeaker_Core/#q19-how-to-change-wake-up-word

closed 2017-10-16 by KillingJacky