respeaker/get_started_with_respeaker #20: how to support google speech ?

xiaooquanwu, 2017-01-16:
how to support google speech ?

xiongyihui, 2017-01-16:
Install speech_recognition library following the guide - https://github.com/respeaker/get_started_with_respeaker/wiki/Use-speech_recognition-python-library

It supports Google Speech API, IBM Speech API and Wit.ai

KillingJacky, 2017-10-16:
Will close this issue, please open a new issue if you have further questions.

closed 2017-10-16 by KillingJacky

tangb, 2018-01-13:
Hi @xiongyihui 

Just a question about this topic: https://github.com/respeaker/get_started_with_respeaker/wiki/Use-speech_recognition-python-library

I want to use speech_recognition api to use google speech but function r.listen() never stops (on this [example](https://github.com/xiongyihui/speech_recognition/blob/respeaker-lite/examples/microphone_recognition.py))

I think everything is correctly installed:
 - I installed all stuff and tested my respeaker 2mic with command "arecord -f cd -Dhw:1 | aplay -Dhw:1" and I got feedback on my speakers
 - I also made some tests with [offline_voice_assistant.py](https://github.com/respeaker/respeaker_python_library/blob/master/examples/offline_voice_assistant.py) example and word 'respeaker' is detected
 - I also running pixel_ring tests and it worked
So I suppose everything works perfectly but using speech_recognition never works.

Can you help me please ?

Thank you very much ;-)

Tanguy

xiongyihui, 2018-01-14:
We can convert the audio generator into a string like:
```
data = mic.listen()
data = b''.join(data)
```

tangb, 2018-01-14:
Thank you for your quick reply
I found another working way based on example I found I don't remember where :p 

```
def convert(audio_data):
    if isinstance(audio_data, types.GeneratorType):
        def generate(audio):
            yield BingSpeechAPI.get_wav_header()
            for a in audio:
                yield a
        data = generate(audio_data)
    else:
        data = BingSpeechAPI.to_wav(audio_data)
    audio = sr.AudioData(''.join(data), 16000, 2)
    return audio

def task():
    ...
    if mic.wakeup('respeaker'):
            print('Wake up')
            data = mic.listen()
            try:
                text = r.recognize_google(convert(data), language='fr-FR')
    ...
```
Thank you again