respeaker/get_started_with_respeaker #25: WiFi setup issues

tangesazen, 2017-01-25:
I found two issues:

1) I have a dash character in the SSID of my access point. When I connect to ReSpeakerXXXXXX, it shows just the part of the text before the dash, and never connects. I setup an access point with no dashes, and was able to select that access point.

2) The new access point had a password with a dollar character '$' and an asterisk '*'. It accepted the password, but wouldn't connect. I looked at /etc/config/wireless, and the password field did not include those two characters. I edited the file by hand to add the characters and was able to connect.

xiongyihui, 2017-01-26:
Thanks for the report. We will track it and try to fix it

tangesazen, 2017-01-27:
To give a bit more detail about #2, I used wictl and was successful at connecting to my router. I was able to ping a location on the internet. I then looked at /etc/config/wireless and found the WiFi password was set, but without the '$' and '*' in it. So on a reboot, I would not be able to connect.

tkurbad, 2017-01-30:
I'm experiencing something similar. Our PSK includes whitespaces. If set via wictl, the password is cut off after the first whitespace character, so I had to manually set it via uci set and uci commit.

isaacjo, 2017-02-12:
Same thing (whitespace in SSID) seen as @tkurbad, can confirm uci set and uci commit, followed by a reboot and my Respeaker happily connects automatically