respeaker/get_started_with_respeaker #26: Flash update gone wrong

Jeroen-H, 2017-01-25:
After flashing update from SD respeaker isn't booting correctly anymore.
See error log below.
Please advise on how to resolve this. Thank you

> [04040C0F][04040D07]
> DDR Calibration DQS reg = 00008989
> 
> 
> UOK
> No initrd
> ## Transferring control to Linux (at address 80000000) ...
> ## Giving linux memsize in MB, 128
> 
> Starting kernel ...
> 
> [    0.000000] Linux version 3.18.23 (pillar@server) (gcc version 4.8.3 (OpenWrt                                                                                                                     /Linaro GCC 4.8-2014.04 r49389) ) #83 Sat Nov 19 07:15:53 PST 2016
> [    0.000000] Board has DDR2
> [    0.000000] Analog PMU set to hw control
> [    0.000000] Digital PMU set to hw control
> [    0.000000] SoC Type: MediaTek MT7688 ver:1 eco:2
> [    0.000000] bootconsole [early0] enabled
> [    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
> [    0.000000] MIPS: machine is Seeed ReSpeaker v1 7688
> [    0.000000] Determined physical RAM map:
> [    0.000000]  memory: 08000000 @ 00000000 (usable)
> [    0.000000] Initrd not found or empty - disabling initrd
> [    0.000000] Zone ranges:
> [    0.000000]   Normal   [mem 0x00000000-0x07ffffff]
> [    0.000000] Movable zone start for each node
> [    0.000000] Early memory node ranges
> [    0.000000]   node   0: [mem 0x00000000-0x07ffffff]
> [    0.000000] Initmem setup node 0 [mem 0x00000000-0x07ffffff]
> [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
> [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 byt                                                                                                                     es
> [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pag                                                                                                                     es: 32512
> [    0.000000] Kernel command line: console=ttyS2,57600 rootfstype=squashfs,jffs                                                                                                                     2
> [    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
> [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
> [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
> [    0.000000] Writing ErrCtl register=0003400f
> [    0.000000] Readback ErrCtl register=0003400f
> [    0.000000] Memory: 126272K/131072K available (2551K kernel code, 128K rwdata                                                                                                                     , 524K rodata, 180K init, 186K bss, 4800K reserved)
> [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
> [    0.000000] NR_IRQS:256
> [    0.000000] intc: using register map from devicetree
> [    0.000000] CPU Clock: 580MHz
> [    0.000000] clocksource_of_init: no matching clocksources found
> [    0.000000] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
> [    0.060000] pid_max: default: 32768 minimum: 301
> [    0.060000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
> [    0.070000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
> [    0.080000] pinctrl core: initialized pinctrl subsystem
> [    0.090000] NET: Registered protocol family 16
> [    0.110000] mt7621_gpio 10000600.gpio: registering 32 gpios
> [    0.120000] mt7621_gpio 10000600.gpio: registering 32 gpios
> [    0.130000] mt7621_gpio 10000600.gpio: registering 32 gpios
> [    0.140000] Switched to clocksource MIPS
> [    0.150000] NET: Registered protocol family 2
> [    0.150000] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
> [    0.170000] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
> [    0.180000] TCP: Hash tables configured (established 1024 bind 1024)
> [    0.190000] TCP: reno registered
> [    0.200000] UDP hash table entries: 256 (order: 0, 4096 bytes)
> [    0.210000] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
> [    0.220000] NET: Registered protocol family 1
> [    0.230000] futex hash table entries: 256 (order: -1, 3072 bytes)
> [    0.250000] squashfs: version 4.0 (2009/01/31) Phillip Lougher
> [    0.260000] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORIT                                                                                                                     Y) (c) 2001-2006 Red Hat, Inc.
> [    0.290000] msgmni has been set to 246
> [    0.290000] io scheduler noop registered
[... 21 lines cut ...]
> [    0.620000] 0x000000050000-0x000000169abb : "kernel"
> [    0.630000] 0x000000169abb-0x000002000000 : "rootfs"
> [    0.640000] mtd: device 5 (rootfs) set to be root filesystem
> [    0.650000] 1 squashfs-split partitions found on MTD device rootfs
> [    0.660000] 0x000001e40000-0x000002000000 : "rootfs_data"
> [    0.690000] ralink_soc_eth 10100000.ethernet eth0: ralink at 0xb0100000, irq                                                                                                                      5
> [    0.700000] mt7621_wdt 10000120.watchdog: Initialized
> [    0.710000] TCP: cubic registered
> [    0.720000] NET: Registered protocol family 17
> [    0.730000] bridge: automatic filtering via arp/ip/ip6tables has been depreca                                                                                                                     ted. Update your scripts to load br_netfilter if you need this.
> [    0.750000] 8021q: 802.1Q VLAN Support v1.8
> [    0.770000] squashfs: SQUASHFS error: unable to read id index table
> [    0.780000] jffs2: Flash size not aligned to erasesize, reducing to 31296KiB
> [    0.790000] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at                                                                                                                      0x00000000: 0x7368 instead
> [    0.810000] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at                                                                                                                      0x00000004: 0x1514 instead
> [    0.830000] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at                                                                                                                      0x00000008: 0x6cc0 instead
> [    0.850000] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at                                                                                                                      0x00000010: 0x00e8 instead
> [    0.870000] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at                                                                                                                      0x00000014: 0x0004 instead
> [    0.890000] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at                                                                                                                      0x00000018: 0x06c0 instead
> [    0.910000] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at                                                                                                                      0x0000001c: 0x0004 instead
> [    0.930000] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at                                                                                                                      0x00000020: 0x05e4 instead
> [    0.940000] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at                                                                                                                      0x00000028: 0x3efa instead
> [    0.960000] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at                                                                                                                      0x00000030: 0x3ef2 instead
> [    0.980000] jffs2: Further such events for this erase block will not be print                                                                                                                     ed
> [    1.000000] jffs2: Empty flash at 0x00000038 ends at 0x00000040

xiongyihui, 2017-01-26:
That's bad. There are two ways to rescue it, but both need a little  bit hardware hack.

1. connect the ReSpeaker's USB D-/D+ pins with a USB storage, then [ upgrade from usb storage](https://docs.labs.mediatek.com/resource/linkit-smart-7688/en/tutorials/firmware-and-bootloader/update-the-firmware-with-a-usb-drive)
2. connect the four Ethernet pins with a Ethernet jack, the upgrade through ftp

Jeroen-H, 2017-01-26:
Hi, ow ok. Will give it a try.
The second option looks the "easiest" to me. Where can I find a pin out of the respeaker?
Have searched the net, but coudn't find one.

Thank you

Wheels05, 2017-01-29:
Did you get any where with this Jeroen I have same issue

Jeroen-H, 2017-01-29:
Hi,

No haven't got the time yet. Possibly Tuesday. What method are you going to
try?

Wheels05, 2017-01-29:
I think the pins that the pins xiongyihui are on about are USBDP, USBDM, VDD_5V and GND on headers J5+J6 and found that if you hold the jog wheel to the left as you plug it in it seems to start a firmware update mode so hopefully just need to connect a USB to the headers and do as above and hey presto. Presume the file to use is the bin file you can download from Github. I presume that is correct or at least seems logical. I’ll be completely honest I’m completely new to all this so unsure hoping you could maybe shed some light on it

Jeroen-H, 2017-01-30:
To be honest, I'm also quite new to all of this. Will do my best in fixing
this and reporting here :)
Where did you find a pinout of the respeaker? I haven't found one :s

Jeroen-H, 2017-01-31:
nevermind found the pinout.
wondering how i'll be connecting this to a usb port now :/

Wheels05, 2017-01-31:
Sorry thought i'd responded to that but cant have done. My plan if/when i get to it was to use some old dupont leads i have and connect them to a front usb connector header off of a pc

Jeroen-H, 2017-02-01:
Ha, that's actually a good idea.
Yesterday I cut the wire of a mouse, but after wiring up and reading everything trough i realized i had to attach a usb rather dan using it as a USB. Gonna check if i can find a usb connector off a pc today.

xiongyihui, 2017-02-06:
The pinout is 
[image]

vlinhd11, 2017-02-08:
I have same issue, try cut off usb cable, wiring up, connect to pins USBDP, USBDM, VDD_5V and GND.
boot up, choose option 5 in loader menu, it say: 
> Register 1111 NbrPorts 1
> USB EHCI 1.00
> scanning bus 0 for devices... cannot reset port 1!?
> 1 USB Device(s) found
>        scanning bus for storage devices... 0 Storage Device(s) found
> No USB Storage found. Upgrade F/W failed.

try to swap USBDP and USBDM check VDD_5V time by time, but it alway say that, now i'm giving up

vlinhd11, 2017-02-10:
just success, forget usb cable, soldering wire into the usb drive, it work like charm
[image]

Jeroen-H, 2017-02-21:
Hi vlindh11, were you able to update the respeaker firmware as well?
I was able to flash the lks img. Then I tried to upgrade to the respeaker firmware from sd card, as mentioned in the how-to's here. But I received the errors in the first post again.

Now I try to reflash the lks img, but the respeaker keeps rebooting.

I also noticed that the respeaker is making squeaking sounds. Has anybody else also noticed this?

xiongyihui, 2017-02-22:
See alsa https://github.com/respeaker/get_started_with_respeaker/wiki/Rescue-from-a-failed-upgrade

Jeroen-H, 2017-02-22:
> wiki/Rescue-from-a-failed-upgrade

Hi,

I was able to get it working by having a lot of patience, reflashing the
firmware mentioned here, a lot of patience again and doing a full format of
my sd card (it appeared to be corrupt).

after each firmware upgrade the respeaker reboots for 5 or 10 minutes, just
wait until it's nice and steady and begin on the following step.

baptistemorand, 2017-03-17:
Hi, 
Ihave the same problem I don't see my usb key where to solder the pin of the usb ? (the picture in the tutorial is not clear)

Jeroen-H, 2017-03-17:
Hi,

I didn't have to solder. I plugged some male solderless breadboard wires
into the device like this

Respeaker - usb
Usbdp - usb+
Usbdm - usb-
Vdd_5v - 5v
Gnd - gnd

The pinout of a usb is 5v, usb+, usb-, gnd if i'm not mistaken.

baptistemorand, 2017-03-17:
thanks for your kick response. I do the same but my usb key is not found ? I will try with an other usb key maybe it's a usb key that is not compatible. (it's a USB of 32G from Verbantin in format FAT32)

Wheels05, 2017-03-17:
I had issues with brand of USB. The one that worked for me was an a plain 8gb Sandisk Cruzer

From: baptistemorand 
Sent: Friday, March 17, 2017 2:20 PM
To: respeaker/get_started_with_respeaker 
Cc: Wheels05 ; Comment 
Subject: Re: [respeaker/get_started_with_respeaker] Flash update gone wrong (#26)

thanks for your kick response. I do the same but my usb key is not found ? I will try with an other usb key maybe it's a usb key that is not compatible. (it's a USB of 32G from Verbantin in format FAT32)

Jeroen-H, 2017-03-17:
Did you attach the usb directly? I used a usb cable from a pc and attached
the wires that are attached to the motherboard to the respeaker.

It can work to attach a usb directly, but be sure not to touch the metal
housing of the usb. It must be connected with gnd.

baptistemorand, 2017-03-17:
I have put male connector like this :
[image]

baptistemorand, 2017-03-17:
I have find my problem. The problem is the ub cable that is too long. I have take a litle usb cable and it work.

KillingJacky, 2017-10-16:
Will close this issue, please open a new issue if you have further questions.

closed 2017-10-16 by KillingJacky

djkmiles, 2018-07-04:
[image]
To add to other info on this, using a shorter USB direct to notebook and using a less noisy memory stick [in pic is 8 GiB] were both necessary. Behaviour was otherwise reading only a few blocks then stopping. Thxxx