respeaker/get_started_with_respeaker #27: /dev/mmcblk0p1 not mounted by fstab

jravetch, 2017-01-26:
I went through the SD expansion instructions and was able to setup the overlay fs; however, the second FAT partition mentioned in the docs is not mounted.

root@ReSpeaker:~# df -h
Filesystem                Size      Used Available Use% Mounted on
rootfs                    5.2G     16.8M      4.9G   0% /
/dev/root                29.0M     29.0M         0 100% /rom
tmpfs                    61.7M      1.1M     60.6M   2% /tmp
/dev/mmcblk0p2            5.2G     16.8M      4.9G   0% /overlay
overlayfs:/overlay        5.2G     16.8M      4.9G   0% /
tmpfs                   512.0K         0    512.0K   0% /dev

Mountd is automounting both partitions to /Media, but this is not what is showing in your example.  Please update your docs.

jerryyip, 2017-02-05:
Hi,
Have you try to mount mmcblk0p1 manually?

AeroFont, 2017-03-31:
Mounting it manually works, but the problem right there is that after some time the mmcblk0p1 partition corrupts itself, which is quite strange.

jacksun101, 2017-04-08:
I am facing the similar issue here.

> 
root@ReSpeaker:~# dmesg | grep fsck
[   40.830000] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.