respeaker/get_started_with_respeaker #29: Not showing up via USB

jtagen, 2017-01-27:
Bad flash from arduino, now no longer being detected as a device over USB.

Any thoughts on cause? Are the ICSP pins broken out?

xiongyihui, 2017-02-06:
We can re-flash the bootloader on openwrt.

```
/etc/init.d/mopidy stop  # stop mopidy if it's running, mopidy-hallo plugin will use SPI
/etc/init.d/alexa stop      # stop alexa if it's running
mt7688_pinmux set ephy gpio
cd /etc/arduino
avrdude -c linuxgpio -p m32u4 -e -U lfuse:w:0xFF:m -U hfuse:w:0xD8:m -U efuse:w:0xCB:m  -U flash:w:Caterina-ReSpeaker.hex -u -U lock:w:0xEF:m
```

jtagen, 2017-02-06:
No serial connection over USB. Are ICSP pins exposed?

xiongyihui, 2017-02-07:
There are 5 test points next to sd card slot, which are ICSP pins. The schematic can be found at https://github.com/respeaker/get_started_with_respeaker/blob/master/Introduction.md#hardware

jtagen, 2017-02-07:
Whew. I can see a lot of test points.... near the SD card 1,3,5,7,8,18-21,23-25,27. 
Schematic doesn't seem to show which ones connect to ICSP. Can you provide more info? Thanks!

qunanan, 2017-02-07:
Have you try using ssh via wifi connection and following the steps as xiongyihui mentioned?

The ICSP pads are:

- SCK     ---->   TP5
- MOSI   ---->   TP7
- MISO   ---->   TP8
- RESET ---->   TP23

[image]

The position for the pads please refer to [RespeakerCorev1.0_PCB_bottom.pdf](https://github.com/respeaker/get_started_with_respeaker/blob/master/files/RespeakerCorev1.0_PCB_bottom.pdf) and [RespeakerCorev1.0_Schematic.pdf](https://github.com/respeaker/get_started_with_respeaker/blob/master/files/RespeakerCorev1.0_Schematic.pdf).

jtagen, 2017-02-07:
Perfect, thanks for the pin info. 

Oddly, WiFi doesn't appear to be active. Definitely not broadcasting as an AP, as it was just before the bad flash. I've checked the DHCP logs on local router (as we as scanned the subnet) and it doesn't appear to be connected. 

Re-reading above after a coffee, @xiongyihui mentions flashing openwrt - this would be for the AI7688. I'm trying to reflash the coprocessor to get back USB/SSH (listed as a LPC11U35 on the schematic, but appears to be a  ATMEGA32U4.

zezokhaled84, 2017-02-21:
guys I'm in the same problem.. I can't understand what to do with the 
The ICSP pads are:

SCK ----> TP5
MOSI ----> TP7
MISO ----> TP8
RESET ----> TP23

im not so expert so if someone can explain to me what to do ?

xiongyihui, 2017-02-23:
It means TP5 in the red square of the picture is SCK.

[image]

SCK ----> TP5
MOSI ----> TP7
MISO ----> TP8
RESET ----> TP23

pastukhov, 2017-03-01:
Hi! I have same problem, but when i'm trying to reflash arduino core  i'm getting:
```
root@ReSpeaker:~# /etc/init.d/mopidy stop  # stop mopidy if it's running, mopidy-hallo plugin will use SPI
root@ReSpeaker:~# /etc/init.d/alexa stop      # stop alexa if it's running
-ash: /etc/init.d/alexa: not found
root@ReSpeaker:~# mt7688_pinmux set ephy gpio
set pinmux ephy -> gpio
root@ReSpeaker:~# cd /etc/arduino
root@ReSpeaker:/etc/arduino# avrdude -c linuxgpio -p m32u4 -e -U lfuse:w:0xFF:m -U hfuse:w:0xD8:m -U efuse:w:0xCB:m  -U flash:w:Caterina-ReSpeaker.hex -u -U lock:w:0xEF:m
Can't export GPIO 15, already exported/busy?: Device or resource busy
avrdude done.  Thank you.
```
Any advice?

baptistemorand, 2017-03-01:
for pastukhov  you can write the cmd :  "echo 15 > /sys/class/gpio/unexport" where 15 is your GPIO.

pastukhov, 2017-03-01:
I'm getting loop:
```
root@ReSpeaker:/etc/arduino# echo 11 > /sys/class/gpio/unexport
root@ReSpeaker:/etc/arduino# avrdude -c linuxgpio -p m32u4 -e -U lfuse:w:0xFF:m -U hfuse:w:0xD8:m -U efuse:w:0xCB:m -U flash:w:Caterina-ReSpeaker.hex -u -U lock:w:0xEF:m 
Can't export GPIO 17, already exported/busy?: Device or resource busy
avrdude done.  Thank you.

root@ReSpeaker:/etc/arduino# echo 17 > /sys/class/gpio/unexport
root@ReSpeaker:/etc/arduino# avrdude -c linuxgpio -p m32u4 -e -U lfuse:w:0xFF:m -U hfuse:w:0xD8:m -U efuse:w:0xCB:m -U flash:w:Caterina-ReSpeaker.hex -u -U lock:w:0xEF:m 
Can't export GPIO 11, already exported/busy?: Device or resource busy
avrdude done.  Thank you.

root@ReSpeaker:/etc/arduino# echo 11 > /sys/class/gpio/unexport
root@ReSpeaker:/etc/arduino# avrdude -c linuxgpio -p m32u4 -e -U lfuse:w:0xFF:m -U hfuse:w:0xD8:m -U efuse:w:0xCB:m -U flash:w:Caterina-ReSpeaker.hex -u -U lock:w:0xEF:m 
Can't export GPIO 15, already exported/busy?: Device or resource busy
avrdude done.  Thank you.

root@ReSpeaker:/etc/arduino# echo 15 > /sys/class/gpio/unexport
root@ReSpeaker:/etc/arduino# avrdude -c linuxgpio -p m32u4 -e -U lfuse:w:0xFF:m -U hfuse:w:0xD8:m -U efuse:w:0xCB:m -U flash:w:Caterina-ReSpeaker.hex -u -U lock:w:0xEF:m 
Can't export GPIO 11, already exported/busy?: Device or resource busy
avrdude done.  Thank you.
```

pastukhov, 2017-03-01:
finally flashed with
```
root@ReSpeaker:/etc/arduino# echo 17 > /sys/class/gpio/unexport
root@ReSpeaker:/etc/arduino# echo 15 > /sys/class/gpio/unexport
root@ReSpeaker:/etc/arduino# echo 11 > /sys/class/gpio/unexport
root@ReSpeaker:/etc/arduino# echo 16 > /sys/class/gpio/unexport
root@ReSpeaker:/etc/arduino# avrdude -c linuxgpio -p m32u4 -e -U lfuse:w:0xFF:m -U hfuse:w:0xD8:m -U efuse:w:0xCB:m -U flash:w:Caterina-ReSpeaker.hex -u -U lock:w:0xEF:m 
```
But got
```
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0x2f != 0xef
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

baptistemorand, 2017-03-01:
Me i am lock my device manager say that I have a "USB IO board" and I can't acess to virtual serial port.

baptistemorand, 2017-03-01:
I have resolve my problem : 
 -> install driver [https://github.com/Seeed-Studio/Signed_USB_Serial_Driver](url)
-> Dowload arduino program [https://github.com/respeaker/get_started_with_respeaker/wiki/Using-Arduino-to-program-ATmega32U4-on-ReSpeaker](url)

AeroFont, 2017-03-28:
What did you flash onto it? Ive got my arduino ready and the pins, but dont really know what to transfer....

baptistemorand, 2017-03-29:
Hi,
See this page :
https://github.com/respeaker/get_started_with_respeaker/blob/master/ProgrammingGuide.md#play-arduino-with-light-touch-sound-and-internet
After you have to custom the code to do what you wnat.
Regards.

*Baptiste Morand.*

*Mobile: 06-35-96-37-97*

*Digital Airways*

*14 rue du Sergent Escoffier*

*61200 Argentan, France*

KillingJacky, 2017-10-16:
About how to recover the ATmega 32u4 bootloader, please refer to https://github.com/respeaker/get_started_with_respeaker/wiki/factory-reset#if-atmega32u4-bootloader-is-broken-

closed 2017-10-16 by KillingJacky