respeaker/get_started_with_respeaker #30: Password for GUI not recognized post-restart

wjhill, 2017-01-28:
## Summary

I changed the password for the router/respeaker GUI, did some more work, restarted and now cannot login to GUI, but can to SSH.

## Background

I changed the password for the GUI, and then went through the [SD Card setup](https://github.com/respeaker/get_started_with_respeaker/blob/master/QuickStart.md#use-sd-card-to-extend-storage). At the end, I did a "reboot" in the terminal.

Upon restart, I login to respeaker via SSH, and realized I don't have Internet connectivity. I go to the GUI to login, and get this error as the page opens:
[image]

I hit OK on the dialog (not shown above), and get this login screen:
[image] 
Neither my old, new, or default root passwords work. (FYI I also changed the IP address, before all the above, and rebooted with that fine.)

## Sum up

How I can:

1. Get myself logged in without resetting? I do have a backup of my settings prior, but if I don't have to [run firstboot](https://github.com/respeaker/get_started_with_respeaker/issues/22#issuecomment-273791785) and understand the issue, I'd be a lot happier.

1. Figure out how to get my respeaker properly on the Internet?

Pillar1989, 2017-02-05:
You can set juci password,"orangectl passwd root ********"

wjhill, 2017-02-06:
Thanks for the response, I will try and respond this week.

closed 2017-02-08 by Pillar1989