respeaker/get_started_with_respeaker #31: Get Multi-channel MIC data thru USB to PC

jeffskl, 2017-02-05:
Can I get multi-channel MIC data with the ReSpeaker Mic Array board thru USB to PC?
Is the MIC data just Raw from the microphone without processed?
Do I need to configure the board to get the Mic raw data? How to do it?

Pillar1989, 2017-02-05:
xmos process the audio data on chip. There are no need to give Mic raw data to user.

jeffskl, 2017-02-05:
I want to get the Mic raw data and then do the signal analysis and algorithm development on PC. So can I get the multi-channel raw data with this board?

Pillar1989, 2017-02-05:
Sorry, you cannt

jeffskl, 2017-02-05:
I see, thank you!

jeffskl, 2017-02-05:
Can I overwrite the FW of the XMOS chip thru JTAG?

Pillar1989, 2017-02-05:
JTAG is very special, you can go to http://www.xmos.com/.

jeffskl, 2017-02-05:
Have Seeed provided the original FW file so that I can revert the function after re-flashing the XMOS chip?

Pillar1989, 2017-02-05:
https://github.com/Fuhua-Chen/ReSpeaker_Microphone_Array_Firmware

jeffskl, 2017-02-05:
That's great. Thanks!

jeffskl, 2017-02-05:
BTW, have the schematic of the Microphone Array Board posted?

Pillar1989, 2017-02-05:
https://drive.google.com/file/d/0BwBIy7Eq4NucRy04LWxHUlJiUUk/view?usp=sharing

doumdi, 2017-02-23:
I am also interested in getting the RAW dat from the array.  I would suggest a firmware update that outputs 9 channels (1,2 = beamformer, 3,4,5,6,7,8,9  = raw output).

eljuligallego, 2017-06-02:
Hello,
I'm insterested also in this topic but I consider this would not be possible due the bottleneck that suppose the usb communication for sending all channels in real time. what do you think?.

also you have to program the xcmos...
http://www.xmos.com/support/examples

xiongyihui, 2017-06-03:
There is a new firmware which supports 8 channels raw data. See https://github.com/respeaker/mic_array_dfu

eljuligallego, 2017-06-03:
thanks for reply xiongyihui,

I see I was mistaken.

chad1023, 2017-07-13:
Excuse me , can you explain more detail about getting respeaker raw data with the new firemware?

chad1023, 2017-07-13:
@xiongyihui 
Excuse me , can you explain more detail about getting respeaker raw data with the new firemware?

xiongyihui, 2017-07-17:
1. use [the dfu tool](https://github.com/respeaker/mic_array_dfu) to upgrade to the new firmware
2. use the python script to read raw audio  data -
 https://github.com/respeaker/mic_array/blob/master/mic_array.py

closed 2017-10-16 by KillingJacky