respeaker/get_started_with_respeaker #32: You do not have an RPC connection to the router!

rakeshsalian, 2017-02-08:
Just opened the pack and connected first time.
Using iPhone Found my ReSpeaker's wifi which i connected...
Selected Wifi and added password...
But when using Laptop i go to the ip 192.168.100.1 using Chrome i get an error window....
[image]

and when clicked ok...
[image]

Also not able to find any documentation explaining the Light indication meanings and use of Navigation Button...

Thanks in advance

Pillar1989, 2017-02-08:
Wow, you are the first user get this error. Can you access to http:192.168.100.1:6680 ? that's a web terminal. Then you will get more debug mesage.

rakeshsalian, 2017-02-08:
Thanks for your reply.
I tried http:192.168.100.1:6680 and i was able to get in and see all files. But again when trying http:192.168.100.1 the above error persists. Can you guide me post http:192.168.100.1:6680 what can i do to get this fixed.

Stebar59, 2017-03-15:
I have exactely the same issue.
Some ideas?

rambocordi, 2017-05-03:
i have also exactly the same issue.

- connecting to 192.168.100.1 (via forefox browser)
- message appears:
uncaught exception: Error: [$rootScope:inprog] http://errors.angularjs.org/1.3.15/$rootScope/inprog?p0=%24digest: 

R/<@http://192.168.100.1/js/01-juci.js:41:5869
p@http://192.168.100.1/js/01-juci.js:42:30911
$apply@http://192.168.100.1/js/01-juci.js:43:2813
fail@http://192.168.100.1/js/01-juci.js:58:17213
j@http://192.168.100.1/js/01-juci.js:29:12694
...
- click OK
- this screen appears
[image]

Connecting to http:192.168.100.1:6680 is NOT working.

Some ideas?

Padamaya, 2018-01-22:
I have the same issue. Any fixes for it?

jdzxd, 2018-01-30:
If you need to connect to the Internet, it is recommended to use the "wictl" command instead of Respeaker's wifi hotspot. Enter "wictl" to connect to the network, then use the "ifconfig" command to see Respeaker's IP address. After networking is successful, log in to Respeaker using SSH and Respeaker's IP address.

Padamaya, 2018-02-04:
Sorry even wictl didn't work. I managed somehow myself. But now, I have the problem, that the respeaker core hooked up with the micarray up crashes all the time (red leds). The whole device is annoying.
Additionally the descriptions in the git hup are irritating. It is difficult to distinguish, if the description are for the micarray hooked up with the respeaker core, or with the raspberry pi. I bought this to test how it is working. If I would have good experience, then I would buy more devices, but so I will keep my hands off the respeaker device. Maybe I wait fro the Matrix Voice or respeaker core V2.