respeaker/get_started_with_respeaker #33: USB Error

PRONOVATEK, 2017-02-09:
Recently I try to execute a Bing python file test.py on respeaker, and always when I try to run, receive the next message: 

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python2.7/site-packages/respeaker/usb_hid/pyusb_backend.py", line 64, in rx_task
    self.rcv_data.append(self.ep_in.read(self.ep_in.wMaxPacketSize, -1))
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 402, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 975, in read
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 216, in setup_request
    self.managed_claim_interface(device, intf)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 818, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 16] Resource busy

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "test.py", line 13, in task
    mic = Microphone(quit_event=quit_event)
  File "/usr/lib/python2.7/site-packages/respeaker/microphone.py", line 67, in __init__
    pixel_ring.set_color(rgb=0x400000)
  File "/usr/lib/python2.7/site-packages/respeaker/pixel_ring.py", line 19, in set_color
    self.write(0, [self.mono_mode, rgb & 0xFF, (rgb >> 8) & 0xFF, (rgb >> 16) & 0xFF])
  File "/usr/lib/python2.7/site-packages/respeaker/pixel_ring.py", line 58, in write
    self.hid.write(packet)
  File "/usr/lib/python2.7/site-packages/respeaker/usb_hid/pyusb_backend.py", line 162, in write
    self.dev.ctrl_transfer(bmRequestType, bmRequest, wValue, wIndex, data)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 1034, in ctrl_transfer
    self._ctx.managed_claim_interface(self, interface_number)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 818, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 16] Resource busy

I reboot the respeaker and try again but always got the same error, any idea?

PRONOVATEK, 2017-02-09:
Now I reset the respeaker with "fistboot" command and try again, after three resets I receive the next verbose, and not run my test.py :

root@ReSpeaker:~# python test.py 
@
INFO:mic:Use ReSpeaker MicArray UAC2.0: USB Audio (hw:1,0)

INFO:mic:Start detecting
INFO:mic:Detected alexa 
INFO:mic:Detected alexa 
INFO:mic:Detected alexa 
INFO:mic:Detected alexa 
INFO:mic:Detected alexa 
INFO:mic:Detected alexa 
INFO:mic:Detected alexa 
INFO:mic:Detected alexa 
INFO:mic:Detected play music 

I don't understand what happening, why execute different options do you have any idea?

regards

xiongyihui, 2017-02-10:
There is a background program `mopidy` which is running and using the audio device. To use your script, firstly stop `mopidy` program by running:

```
/etc/init.d/mopidy/stop
/etc/init.d/mopidy/disable
```

PRONOVATEK, 2017-02-10:
Thanks, that resolve my problems

closed 2017-02-10 by PRONOVATEK