respeaker/get_started_with_respeaker #35: Noise cancelling

elthef, 2017-02-10:
Just to find does the microphone array removed background noise?
I also wanted to find out how to get final beamforming data.

elpimous, 2017-02-13:
Hello, i'd like to hear too, an audio file with raw sentence before/after dsp
To hear efficiency of this dsp mic
Please
Vincent

Fuhua-Chen, 2017-02-17:
@elpimous  @elthef 
It seems caused by AGC，please use the [HID-TOOL](https://github.com/Fuhua-Chen/ReSpeaker-Microphone-Array-HID-tool) to find the root cause.
You can close the AGC and set a low gain.
beamforming data also can be got by HID-TOOL
On respeaker core, also can try by this [](https://github.com/respeaker/get_started_with_respeaker/wiki/Mic-Array)