respeaker/get_started_with_respeaker #36: Pygame Install Error

PRONOVATEK, 2017-02-10:
Hi, recently I tried to install a Pygame with command "pip install pygame" and receive the next error:

root@ReSpeaker:~# pip install pygame
Downloading/unpacking pygame
  Downloading pygame-1.9.3.tar.gz (3.0MB): 3.0MB downloaded
  Running setup.py (path:/tmp/pip_build_root/pygame/setup.py) egg_info for package pygame
root@ReSpeaker:~# pip install pygame
Downloading/unpacking pygame
  Downloading pygame-1.9.3.tar.gz (3.0MB): 3.0MB downloaded
  Running setup.py (path:/tmp/pip_build_root/pygame/setup.py) egg_info for package pygame
    
    
    WARNING, No "Setup" File Exists, Running "config.py"
    Using UNIX configuration...
    
    sh: sdl-config: not found
    sh: sdl-config: not found
    sh: sdl-config: not found
    sh: freetype-config: not found
    sh: freetype-config: not found
    sh: freetype-config: not found
    
    Hunting dependencies...
    WARNING: "sdl-config" failed!
    WARNING: "freetype-config" failed!
    Unable to run "sdl-config". Please make sure a development version of SDL is installed.
    Complete output from command python setup.py egg_info:
    

WARNING, No "Setup" File Exists, Running "config.py"

Using UNIX configuration...

sh: sdl-config: not found

sh: sdl-config: not found

sh: sdl-config: not found

sh: freetype-config: not found

sh: freetype-config: not found

sh: freetype-config: not found

Hunting dependencies...

WARNING: "sdl-config" failed!

WARNING: "freetype-config" failed!

Unable to run "sdl-config". Please make sure a development version of SDL is installed.

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip_build_root/pygame
Storing debug log for failure in /root/.pip/pip.log

I understand that S.O. is OpenWRT, are any other method to install python-pygame into this version?

regards

xiongyihui, 2017-02-13:
It seems PyGame requires cross-compile on OpenWrt and couldn't be installed through pip.  If just using pygame to record/play audio, we'd better use PyAudio.

It's not easy to build PyGame for OpenWrt. If you want to try, the following links may help:

1. https://wiki.openwrt.org/doc/howtobuild/single.package
2. https://github.com/jameshilliard/openwrt-packages/tree/master/pygame

PRONOVATEK, 2017-02-13:
ok, perfect. Thanks a lot for your support

closed 2017-02-13 by PRONOVATEK