respeaker/get_started_with_respeaker #38: panic

xiaooquanwu, 2017-02-13:
i have compiled the source and upgraded the Respeaker,it panic ! can somebody help to find what the issue is ?
```
[   11.630000] jffs2: notice: (377) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[   11.670000] block: attempting to load /tmp/jffs_cfg/upper/etc/config/fstab
[   11.680000] block: extroot: device not present, retrying in 1 seconds
[   12.710000] block: extroot: cannot find device mmcblk0p2
[   12.790000] jffs2: notice: (374) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[   12.820000] mount_root: loading kmods from internal overlay
[   13.110000] block: attempting to load /tmp/jffs_cfg/upper/etc/config/fstab
[   13.120000] block: extroot: device not present, retrying in 1 seconds
[   14.150000] block: extroot: cannot find device mmcblk0p2
[   14.170000] mount_root: switching to jffs2 overlay
[   14.200000] procd: - early -
[   14.210000] procd: - watchdog -
[   15.020000] procd: - ubus -
[   16.040000] procd: - init -
Please press Enter to activate this console.
[   18.390000] NET: Registered protocol family 10
[   18.460000] RPC: Registered named UNIX socket transport module.
[   18.480000] RPC: Registered udp transport module.
[   18.490000] RPC: Registered tcp transport module.
[   18.500000] RPC: Registered tcp NFSv4.1 backchannel transport module.
[   18.520000] ntfs: driver 2.1.31 [Flags: R/O MODULE].
[   18.610000] ip6_tables: (C) 2000-2006 Netfilter Core Team
[   18.630000] i2c /dev entries driver
[   19.290000] <-- RTMPAllocTxRxRingMemory, Status=0, ErrorValue=0x
[   19.310000] <-- RTMPAllocAdapterBlock, Status=0
[   19.320000] mtk_gpio_wifi gpio-wifi: registering 1 gpio
[   19.360000] i2c-ralink 10000900.i2c: loaded
[   19.380000] Linux video capture interface: v2.00
[   19.420000] hidraw: raw HID events driver (C) Jiri Kosina
[   19.450000] fuse init (API version 7.23)
[   19.460000] Enable Ralink GDMA Controller Module 
[   19.470000] GDMA IP Version=3
[   19.640000] *******Enter codec_wm8960_i2c_probe********
[   19.650000] audio gpio number: 41
[   19.650000] audio gpio number: 12
[   19.660000] soc-audio soc-audio: ASoC: machine MTK APSoC I2S should use snd_soc_register_card()
[   19.680000] wm8960 0-001a: No platform data supplied
[   19.690000] ****** wm8960_preinit ******
[   20.200000] soc-audio soc-audio: wm8960-hifi <-> mt76xx-i2s mapping ok
[   20.220000] ip_tables: (C) 2000-2006 Netfilter Core Team
[   20.270000] nf_conntrack version 0.5.0 (1975 buckets, 7900 max)
[   20.360000] hid-generic 0003:2886:0007.0001: hidraw0: USB HID v1.10 Device [SeeedStudio ReSpeaker MicArray UAC2.0] on usb-101c1000.ohci-1/input4
[   20.390000] usbcore: registered new interface driver usbhid
[   20.400000] usbhid: USB HID core driver
[   20.420000] usbcore: registered new interface driver uvcvideo
[   20.430000] USB Video Class driver (1.1.1)
[   20.470000] xt_time: kernel timezone is -0000
[   20.490000] PPP generic driver version 2.4.2
[   20.500000] NET: Registered protocol family 24
[   21.450000] usbcore: registered new interface driver snd-usb-audio
[   29.660000] device eth0.1 entered promiscuous mode
[   29.670000] device eth0 entered promiscuous mode
[   29.690000] br-lan: port 1(eth0.1) entered forwarding state
[   29.700000] br-lan: port 1(eth0.1) entered forwarding state
[   31.170000] efuse_probe: efuse = 10000012
[   31.390000] tssi_0_target_pwr_g_band = 34
[   31.400000] tssi_1_target_pwr_g_band = 35
[   31.700000] br-lan: port 1(eth0.1) entered forwarding state
[   38.530000] random: nonblocking pool is initialized
[   39.060000] <==== rt28xx_init, Status=0
[   40.670000] device ra0 entered promiscuous mode
[   40.680000] br-lan: port 2(ra0) entered forwarding state
[   40.690000] br-lan: port 2(ra0) entered forwarding state
[   42.690000] br-lan: port 2(ra0) entered forwarding state
[   43.080000] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
autorun
[   93.520000] capture free_dma_buffer
[   93.530000] ptri2s_config->mmap_index:0
[... 60 lines cut ...]
[   99.820000] Process python (pid: 1660, threadinfo=862d6000, task=862b3a20, tls=770ee750)
[   99.820000] Stack : 878289b0 8003d014 ffffffff 00000000 878289b0 00000000 80310000 00000000
          000002d2 8003d390 862d6000 862d7c18 000002d2 00000000 8030a640 00000000
          000000d2 00000000 80309570 80026f0c 00000010 800713ec 00000000 00000000
          00000001 8004dde4 00000000 00000141 8030a100 00000000 00200000 8000a198
          00000000 00000000 8030a100 00000022 00200000 000000d2 00000000 863a7980
          ...
[   99.820000] Call Trace:
[   99.820000] [<8013a338>] rb_insert_color+0x2c/0x14c
[   99.820000] [<80040f64>] enqueue_task_fair+0x134/0x174
[   99.820000] [<8003d390>] try_to_wake_up+0x58/0x130
[   99.820000] [<80026f0c>] raise_softirq+0x40/0x60
[   99.820000] [<8004dde4>] rcu_sched_qs+0x8c/0xac
[   99.820000] [<8000a198>] __schedule+0x40/0x5c0
[   99.820000] [<8003d4e0>] __cond_resched+0x20/0x3c
[   99.820000] [<8000a930>] _cond_resched+0x2c/0x3c
[   99.820000] [<80095358>] __vmalloc_node_range+0x1b0/0x29c
[   99.820000] [<80095474>] __vmalloc_node+0x30/0x3c
[   99.820000] [<80095520>] vmalloc+0xa0/0xac
[   99.820000] [<86f34f1c>] snd_pcm_plugin_client_channels+0x8c/0x1ac [snd_pcm_oss]
[   99.820000] 
[   99.820000] 
Code: 30660001  14c00047  00000000 <8c660004> 10460016  00000000  10c00005  00000000  8cc80000 
[  100.510000] ---[ end trace 600de7407b02abb4 ]---
```

xiongyihui, 2017-02-13:
Is it an error produced by mopidy?  run `/etc/init.d/mopidy stop` to disable it. Could you be able to run arecord or aplay?

xiaooquanwu, 2017-02-13:
yes,when disable mopidy or alexa, panic disappear, aplay works ok

xiongyihui, 2017-02-13:
If arecord works but alexa does't, it may be a compiling issue of portaudio. We can test it with pyaudio example code

```
"""PyAudio example: Record a few seconds of audio and save to a WAVE file."""

import pyaudio
import wave

CHUNK = 1024
FORMAT = pyaudio.paInt16
CHANNELS = 2
RATE = 44100
RECORD_SECONDS = 5
WAVE_OUTPUT_FILENAME = "output.wav"

p = pyaudio.PyAudio()

stream = p.open(format=FORMAT,
                channels=CHANNELS,
                rate=RATE,
                input=True,
                frames_per_buffer=CHUNK)

print("* recording")

frames = []

for i in range(0, int(RATE / CHUNK * RECORD_SECONDS)):
    data = stream.read(CHUNK)
    frames.append(data)

print("* done recording")

stream.stop_stream()
stream.close()
p.terminate()

wf = wave.open(WAVE_OUTPUT_FILENAME, 'wb')
wf.setnchannels(CHANNELS)
wf.setsampwidth(p.get_sample_size(FORMAT))
wf.setframerate(RATE)
wf.writeframes(b''.join(frames))
wf.close()
```

xiaooquanwu, 2017-02-13:
```
root@ReSpeaker:/bin# python pyaudio.py 
Traceback (most recent call last):
  File "pyaudio.py", line 3, in <module>
    import pyaudio
  File "/bin/pyaudio.py", line 7, in <module>
    FORMAT = pyaudio.paInt16
AttributeError: 'module' object has no attribute 'paInt16'
```

xiaooquanwu, 2017-02-13:
a compiling issue of portaudio ? how to  fix ?

xiongyihui, 2017-02-13:
You should rename `pyaudio.py`, `pyaudio` is library name and should not be used as your script name.

xiaooquanwu, 2017-02-13:
```
root@ReSpeaker:/bin# python pyaudio_example.py 
[ 1548.650000] capture free_dma_buffer
[ 1548.660000] ptri2s_config->mmap_index:0
[ 1548.670000] playback free_dma_buffer
[ 1548.680000] ptri2s_config->mmap_index:0
[ 1548.870000] capture free_dma_buffer
[ 1548.880000] ptri2s_config->mmap_index:0
[ 1548.890000] playback free_dma_buffer
[ 1548.900000] ptri2s_config->mmap_index:0
[ 1548.920000] capture free_dma_buffer
[ 1548.930000] ptri2s_config->mmap_index:0
[ 1548.940000] playback free_dma_buffer
[ 1548.950000] ptri2s_config->mmap_index:0
[ 1549.220000] ****** wm8960_postinit ******
[ 1549.240000] playback free_dma_buffer
[ 1549.250000] ptri2s_config->mmap_index:0
[ 1549.320000] ummap MMAP[0]=0x85F20000
[ 1549.330000] playback free_dma_buffer
[ 1549.340000] ptri2s_config->mmap_index:8
[ 1549.610000] ****** wm8960_postinit ******
[ 1549.890000] ****** wm8960_postinit ******
[ 1550.170000] ****** wm8960_postinit ******
[ 1550.450000] ****** wm8960_postinit ******
[ 1550.730000] ****** wm8960_postinit ******
[ 1551.010000] ****** wm8960_postinit ******
[ 1551.290000] ****** wm8960_postinit ******
[ 1551.570000] ****** wm8960_postinit ******
[ 1551.850000] ****** wm8960_postinit ******
[ 1552.130000] ****** wm8960_postinit ******
[ 1552.410000] ****** wm8960_postinit ******
[ 1552.690000] ****** wm8960_postinit ******
[ 1552.970000] ****** wm8960_postinit ******
[ 1553.250000] ****** wm8960_postinit ******
[ 1553.530000] ****** wm8960_postinit ******
[ 1553.820000] ****** wm8960_postinit ******
[ 1554.100000] ****** wm8960_postinit ******
[ 1554.120000]  mtk_dai: ASoC: unmatched rate symmetry: 8000 - 44100
[ 1554.130000] capture free_dma_buffer
[ 1554.140000] ptri2s_config->mmap_index:8
Expression 'ioctl( devHandle, SNDCTL_DSP_SPEED, &sr )' failed in 'src/hostapi/oss/pa_unix_oss.c', line: 421
[ 1554.410000] ****** wm8960_postinit ******
[ 1554.430000] ****** wm8960_preinit ******
[ 1554.950000] Kernel bug detected[#1]:
[ 1554.950000] CPU: 0 PID: 2038 Comm: python Not tainted 3.18.23 #2
[ 1554.950000] task: 85ee5d10 ti: 85d34000 task.ti: 85d34000
[ 1554.950000] $ 0   : 00000000 80008b84 00000001 00000b20
[ 1554.950000] $ 4   : 00000000 00000020 00000020 ffffffbf
[ 1554.950000] $ 8   : 0020001e 85cdc700 00000020 0020001e
[ 1554.950000] $12   : 00000000 00000001 00000000 00000000
[ 1554.950000] $16   : 85c84880 00000000 00000001 85ce3c00
[ 1554.950000] $20   : 878b9b80 00000000 00001f40 00000000
[ 1554.950000] $24   : 00000000 86f29ae8                  
[ 1554.950000] $28   : 85d34000 85d35d30 00001f40 80094e9c
[ 1554.950000] Hi    : 00000000
[ 1554.950000] Lo    : 00000000
[ 1554.950000] epc   : 80094efc __vunmap+0xa4/0x114
[ 1554.950000]     Not tainted
[ 1554.950000] ra    : 80094e9c __vunmap+0x44/0x114
[ 1554.950000] Status: 1100e403 KERNEL EXL IE 
[ 1554.950000] Cause : 10800034
[ 1554.950000] PrId  : 00019655 (MIPS 24KEc)
[ 1554.950000] Modules linked in: pppoe ppp_async iptable_nat snd_usb_audio pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT uvcvideo usbhid spi_gpio spi_bitbang snd_usbmidi_lib slhc nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_ftp nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables hid_generic crc_ccitt snd_soc_mt76xx_machine snd_soc_mt76xx_pcm snd_soc_mt76xx_i2s snd_soc_mt76xx_i2s_ctl i2c_wm8960 snd_soc_wm8960 ralink_gdma fuse videobuf2_vmalloc videobuf2_memops videobuf2_core hid v4l2_common videodev evdev i2c_ralink snd_soc_core mt_wifi i2c_dev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables nfsv3 nfs snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi snd_seq_device snd_hwdep snd input_core soundcore vfat fat ntfs lockd sunrpc grace cifs autofs4 nls_utf8 nls_iso8859_2 nls_iso8859_15 nls_iso8859_1 nls_cp437 regmap_spi regmap_i2c i2c_core regmap_core lzo_decompress lzo_compress ipv6 dma_shared_buffer sha256_generic md5 md4 hmac ecb des_generic arc4 crypto_blkcipher mmc_block usb_storage sdhci_pltfm sdhci mtk_sd mmc_core leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common crc16 aead crypto_hash
[ 1554.950000] Process python (pid: 2038, threadinfo=85d34000, task=85ee5d10, tls=7760b750)
[ 1554.950000] Stack : 00000002 85ce2800 00103fff 0020001d 85cdc800 85cdc900 86d9f000 86f35150
          00000000 00000000 00001f40 00001f40 86d9f000 86f314a8 85ce3800 8643c000
          86d9f000 85ce3c00 878b9b80 85ce3800 8643c000 86f324b8 fffffff6 85d35e60
          00000000 00000000 00000004 00000000 00000000 00000000 00000000 00000000
          00000000 00000000 00000018 00000000 00000000 00000000 00000000 00000000
          ...
[ 1554.950000] Call Trace:
[ 1554.950000] [<80094efc>] __vunmap+0xa4/0x114
[ 1554.950000] [<86f35150>] snd_pcm_plugin_free+0x38/0x54 [snd_pcm_oss]
[ 1554.950000] [<86f314a8>] 0x86f314a8
[ 1554.950000] 
[ 1554.950000] 
Code: 00822021  8c840000  2c820001 <00020336> 0c01c195  00002821  080253b7  26310001  8e02000c 
[ 1555.530000] ---[ end trace 9580967151fb5a56 ]---
```

xiongyihui, 2017-02-13:
It seems you include oss module support for portaudio, you should disable oss in your portaudio's makefile

xiaooquanwu, 2017-02-13:
disable oss in  portaudio's makefile, it still panic 

```
root@ReSpeaker:/bin# python pyaudio_example.py 
[   85.410000] capture free_dma_buffer
[   85.420000] ptri2s_config->mmap_index:0
[   85.430000] playback free_dma_buffer
[   85.430000] ptri2s_config->mmap_index:0
[   85.700000] capture free_dma_buffer
[   85.710000] ptri2s_config->mmap_index:0
[   85.740000] playback free_dma_buffer
[   85.750000] ptri2s_config->mmap_index:0
[   85.780000] capture free_dma_buffer
[   85.790000] ptri2s_config->mmap_index:0
[   85.810000] playback free_dma_buffer
[   85.820000] ptri2s_config->mmap_index:0
[   86.110000] ****** wm8960_postinit ******
[   86.130000] playback free_dma_buffer
[   86.140000] ptri2s_config->mmap_index:0
[   86.250000] ummap MMAP[0]=0x86370000
[   86.250000] playback free_dma_buffer
[   86.260000] ptri2s_config->mmap_index:8
[   86.620000] ****** wm8960_postinit ******
[   86.860000] CPU 0 Unable to handle kernel paging request at virtual address 00000004, epc == 8013a338, ra == 8013b6d4
[   86.860000] Oops[#1]:
[   86.860000] CPU: 0 PID: 1268 Comm: odhcpd Not tainted 3.18.23 #3
[   86.860000] task: 879b3a20 ti: 86036000 task.ti: 86036000
[   86.860000] $ 0   : 00000000 00000000 862cbde0 00000000
[   86.860000] $ 4   : 86037e68 803099f4 00000000 fffffffc
[   86.860000] $ 8   : b0bd76a1 00000000 001e847f 00000000
[   86.860000] $12   : 00000014 776bc3a0 77725658 7772c000
[   86.860000] $16   : 86037e68 803099f4 86037e68 803099e8
[   86.860000] $20   : 00000001 00000000 0000000a 87b02500
[   86.860000] $24   : 000003e8 7769d260                  
[   86.860000] $28   : 86036000 86037db8 86d0201c 8013b6d4
[   86.860000] Hi    : 00000014
[   86.860000] Lo    : b09ef222
[   86.860000] epc   : 8013a338 rb_insert_color+0x2c/0x14c
[   86.860000]     Not tainted
[   86.860000] ra    : 8013b6d4 timerqueue_add+0xc0/0x118
[   86.860000] Status: 1100e402 KERNEL EXL 
[   86.860000] Cause : 00800008
[   86.860000] BadVA : 00000004
[   86.860000] PrId  : 00019655 (MIPS 24KEc)
[   86.860000] Modules linked in: pppoe ppp_async iptable_nat snd_usb_audio pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT uvcvideo usbhid spi_gpio spi_bitbang snd_usbmidi_lib slhc nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_ftp nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables hid_generic crc_ccitt snd_soc_mt76xx_machine snd_soc_mt76xx_pcm snd_soc_mt76xx_i2s snd_soc_mt76xx_i2s_ctl i2c_wm8960 snd_soc_wm8960 ralink_gdma fuse videobuf2_vmalloc videobuf2_memops videobuf2_core hid v4l2_common videodev evdev i2c_ralink snd_soc_core mt_wifi i2c_dev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables nfsv3 nfs snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi snd_seq_device snd_hwdep snd input_core soundcore vfat fat ntfs lockd sunrpc grace cifs autofs4 nls_utf8 nls_iso8859_2 nls_iso8859_15 nls_iso8859_1 nls_cp437 regmap_spi regmap_i2c i2c_core regmap_core lzo_decompress lzo_compress ipv6 dma_shared_buffer sha256_generic md5 md4 hmac ecb des_generic arc4 crypto_blkcipher mmc_block usb_storage sdhci_pltfm sdhci mtk_sd mmc_core leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common crc16 aead crypto_hash
[   86.860000] Process odhcpd (pid: 1268, threadinfo=86036000, task=879b3a20, tls=77741440)
[   86.860000] Stack : 00000000 00000000 00000000 00000000 b09ef222 00000014 86037e68 80050534
          879b3a50 80309570 80310000 862b4508 001e847f 00000000 b0bd76a1 00000014
          000080fe 879b3a20 86037ef0 00000000 001e847f 87b02500 001e847f 777255b0
          0000000a 87b02500 86d0201c 800506e8 86037ef0 00000000 86037e68 800502bc
          001e847f 00000000 00000001 00000000 001e847f 8000cb64 00000000 00002019
          ...
[   86.860000] Call Trace:
[   86.860000] [<8013a338>] rb_insert_color+0x2c/0x14c
[   86.860000] [<8013b6d4>] timerqueue_add+0xc0/0x118
[   86.860000] [<80050534>] __hrtimer_start_range_ns+0x100/0x290
[   86.860000] [<800506e8>] hrtimer_start_range_ns+0x24/0x30
[   86.860000] [<8000cb64>] schedule_hrtimeout_range_clock+0xc0/0x118
[   86.860000] [<800d5b40>] SyS_epoll_wait+0x324/0x370
[   86.860000] [<80006b5c>] handle_sys+0x11c/0x140
[   86.860000] 
[   86.860000] 
Code: 30660001  14c00047  00000000 <8c660004> 10460016  00000000  10c00005  00000000  8cc80000 
[   87.510000] ---[ end trace 828ce27336de8338 ]---
[   87.530000] ****** wm8960_postinit ******
[   87.540000] CPU 0 Unable to handle kernel paging request at virtual address 00000028, epc == 801095b0, ra == 801095b0
[   87.540000] Oops[#2]:
[   87.540000] CPU: 0 PID: 1525 Comm: python Tainted: G      D        3.18.23 #3
[   87.540000] task: 861b9838 ti: 861d2000 task.ti: 861d2000
[   87.540000] $ 0   : 00000000 779c35e8 00000000 00000001
[... 65 lines cut ...]
          00000014 9cd19574 00000014 00000001 8030cdd4 00000007 00000000 00000000
          ...
[   87.540000] Call Trace:
[   87.540000] [<8013a6f4>] rb_erase+0x29c/0x3cc
[   87.540000] [<8013b7b0>] timerqueue_del+0x6c/0x88
[   87.540000] [<80050178>] __remove_hrtimer+0x48/0xf8
[   87.540000] [<80050388>] __run_hrtimer.isra.28+0x4c/0xf8
[... cut ...]
[   87.540000] [<800110b4>] r4k_wait_irqoff+0x18/0x20
[   87.540000] [<80043540>] cpu_startup_entry+0xf0/0x13c
[   87.540000] [<80323b5c>] start_kernel+0x42c/0x444
[   87.540000] 
[   87.540000] 
Code: 34470001  ac440008  ac620004 <ac870000> 8c480000  01063824  ac680000  10e00008  ac430000 
[   87.540000] ---[ end trace 828ce27336de833a ]---
[   87.540000] Kernel panic - not syncing: Fatal exception in interrupt
[   87.540000] Rebooting in 3 seconds..[04030C0E][04030C0D]
DDR Calibration DQS reg = 00008888
```