respeaker/get_started_with_respeaker #39: Sorry i can't run the playmusic.py....

calvinwongsc, 2017-02-13:
Hi,

I have a BING_KEY insert to  playmusic.py. but it return the following result.... how can i get it work?
"Try `madplay --help' for more information.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python2.7/site-packages/respeaker/usb_hid/pyusb_backend.py", line 64, in rx_task
    self.rcv_data.append(self.ep_in.read(self.ep_in.wMaxPacketSize, -1))
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 402, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 975, in read
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 216, in setup_request
    self.managed_claim_interface(device, intf)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 818, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 16] Resource busy

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "playmusic.py", line 16, in task
    mic = Microphone(quit_event=quit_event)
  File "/usr/lib/python2.7/site-packages/respeaker/microphone.py", line 67, in __init__
    pixel_ring.set_color(rgb=0x400000)
  File "/usr/lib/python2.7/site-packages/respeaker/pixel_ring.py", line 19, in set_color
    self.write(0, [self.mono_mode, rgb & 0xFF, (rgb >> 8) & 0xFF, (rgb >> 16) & 0xFF])
  File "/usr/lib/python2.7/site-packages/respeaker/pixel_ring.py", line 58, in write
    self.hid.write(packet)
  File "/usr/lib/python2.7/site-packages/respeaker/usb_hid/pyusb_backend.py", line 162, in write
    self.dev.ctrl_transfer(bmRequestType, bmRequest, wValue, wIndex, data)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 1034, in ctrl_transfer
    self._ctx.managed_claim_interface(self, interface_number)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 818, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 16] Resource busy"

jimnet2005, 2017-02-14:
Same for me. I was able to play music a couple of times, but now the device is getting the error below. Could you please help?

root@ReSpeaker:~# python playmusic.py
Usage: madplay [OPTIONS] FILE [...]
Try `madplay --help' for more information.
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "playmusic.py", line 16, in task
    mic = Microphone(quit_event=quit_event)
  File "/usr/lib/python2.7/site-packages/respeaker/microphone.py", line 67, in __init__
    pixel_ring.set_color(rgb=0x400000)
  File "/usr/lib/python2.7/site-packages/respeaker/pixel_ring.py", line 36, in set_color
    self.write(0, [self.mono_mode, rgb & 0xFF, (rgb >> 8) & 0xFF, (rgb >> 16) & 0xFF])
  File "/usr/lib/python2.7/site-packages/respeaker/pixel_ring.py", line 75, in write
    self.hid.write(packet)
  File "/usr/lib/python2.7/site-packages/respeaker/usb_hid/pyusb_backend.py", line 162, in write
    self.dev.ctrl_transfer(bmRequestType, bmRequest, wValue, wIndex, data)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 1034, in ctrl_transfer
    self._ctx.managed_claim_interface(self, interface_number)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 818, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 16] Resource busy

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python2.7/site-packages/respeaker/usb_hid/pyusb_backend.py", line 64, in rx_task
    self.rcv_data.append(self.ep_in.read(self.ep_in.wMaxPacketSize, -1))
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 402, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 975, in read
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 216, in setup_request
    self.managed_claim_interface(device, intf)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 818, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python2.7/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 16] Resource busy

hengway1989, 2017-02-14:
I also having the exactly same issue like above, please help. Thank you

xiongyihui, 2017-02-14:
It should be that `mopidy` is running in background and is using the USB device. try to run `/etc/init.d/mopidy stop` to stop mopidy and run your command again.

jimnet2005, 2017-02-14:
Yes. The command above solved the issue for me. Now the next issue is that ReSpeaker only recognizes the wake up command, but not the "play music" command. Immediately after saying "respeaker" the hardware enters a state in which the leds are showing a circular green light pattern, and the only command recognized seems to be "alexa" (and I haven't said that). I'm pronouncing play music even with Google translate with no result. After several attempts, the speaker printed "Detected play music" at certain distance from the speaker (2 meters or so)

root@ReSpeaker:~# python playmusic.py
Usage:madplay [OPTIONS] FILE [...]
Try `madplay --help' for more information.
@
[ 1044.260000] capture free_dma_buffer
[ 1044.270000] ptri2s_config->mmap_index:8
[ 1044.280000] playback free_dma_buffer
[ 1044.290000] ptri2s_config->mmap_index:8
[ 1044.340000] capture free_dma_buffer
[ 1044.340000] ptri2s_config->mmap_index:8
[ 1044.360000] playback free_dma_buffer
[ 1044.370000] ptri2s_config->mmap_index:8
[ 1044.390000] capture free_dma_buffer
[ 1044.400000] ptri2s_config->mmap_index:8
[ 1044.410000] playback free_dma_buffer
[ 1044.420000] ptri2s_config->mmap_index:8
[ 1044.700000] ****** wm8960_postinit ******
[ 1044.720000] playback free_dma_buffer
[ 1044.730000] ptri2s_config->mmap_index:8
[ 1044.800000] ummap MMAP[0]=0x81810000
[ 1044.810000] playback free_dma_buffer
[ 1044.820000] ptri2s_config->mmap_index:8
INFO:mic:Use ReSpeaker MicArray UAC2.0: USB Audio (hw:1,0)

INFO:mic:Start detecting
INFO:mic:Detected respeaker
Wake up

INFO:mic:Start listening
INFO:requests.packages.urllib3.connectionpool:Starting new HTTPS connection (1): oxford-speech.cloudapp.net
DEBUG:requests.packages.urllib3.connectionpool:"POST /token/issueToken HTTP/1.1" 401 143
1111+111111111111111111111111111111111111111111111111111001recognition connection failed

INFO:mic:Start detecting
11111110000001111111100000000-00000000000000000000000000000001111+00000000-000000000000000000000000000000000000000000000000000000000000
INFO:mic:Stop listening
INFO:micDetected alexa
INFO:mic:Detected alexa
INFO:mic:Detected alexa
INFO:mic:Detected play music
INFO:mic:Detected alexa
INFO:mic:Detected alexa

Is there any other settings we need to check? . Note the "recognition connection failed error" above. This seems to be random as well.

jimnet2005, 2017-02-14:
UPDATE: answer seems to be here: https://github.com/respeaker/get_started_with_respeaker/issues/13. Now I have to figure out how to "git clone" in OpenWrt. If anyone is out there to advise in the meantime I would appreciate t.

jimnet2005, 2017-02-14:
To install git:
opkg update
opkg install git
But then I got an error saying there is no space on device. Any suggestions for extending the filesystem to the flash drive?

calvinwongsc, 2017-02-15:
i am using the web terminal to type /etc/init.d/mopidy disable
but it said

rm: can't remove '/etc/rc.d/S80mopidy': Read-only file system
rm: can't remove '/etc/rc.d/K89mopidy': Read-only file system

so i can't update my python also....

quezbeme, 2017-02-16:
On step 4 I get "Illegal instruction"
Any idea why I get that?

quezbeme, 2017-02-18:
ok I eventually I got it to work. 
First I followed through all the instructions listed above. 
Then I put playmusic.py in the respeaker_python_library/respeaker folder. Then I ran the terminal via ssh (serial and web wouldn't quite work. I assume cos of the USB conflict with mopidy even after stop/disable and the web term being based off mopidy I think?). 
Ran /etc/init.d/mopidy stop
cd to the folder playmusic.py is in and ran python playmusic.py
The first time I ran it I got an error. Second time I ran it I got it to work. What I noticed is you can call "respeaker" loudly from a distance but you need to be closer for it to recognize when you tell it to "play music."
Hope that helps any

aleifuu, 2017-02-20:
Like @quezbeme , I have to copy my code to respeaker_python_library in order to make it work ... otherwise it will simply put an error saying can't find microphone class 

Isn't this command `python setup.py install` supposed to get the clone git repository into the library ? hence you can store and call your code from /root ( or home ) directory ? I have not figured out how to get it installed properly into library I guess

Any pointer ?

KillingJacky, 2017-10-17:
I suppose @xiongyihui 's answer has addressed the original issue. About `setup.py`, please refer to https://stackoverflow.com/questions/1471994/what-is-setup-py

closed 2017-10-17 by KillingJacky

mimansamaheshwari, 2017-11-26:
I am using Respeaker Core that uses chaos calmer v0.9.99.
I am unable to run any  python file on respeaker. I tried to resolve the problem by the ways listed above but in that also I am not able to install  'respeaker_python_library' . 
I used the following command and got this:
```
root@ReSpeaker:~# install git clone https://github.com/respeaker/respeaker_python_library.git
-ash: install: not found
```
Is there anything wrong with this command ?
 Are there any other steps needed for configuring respeaker properly so that I can run python codes?
Is there any need to update system?

KillingJacky, 2017-11-27:
@mimansamaheshwari your command doesn't make sense. 

The `respeaker_python_library` is by default installed on the ReSpeaker Core. I assume that you use the system image for driving the MeowKing Drive Unit in which system we striped some libraries to save space and lower down the CPU load. Please refer to https://github.com/respeaker/get_started_with_respeaker/blob/master/docs/ReSpeaker/QuickStart.md#respeaker-drive-unit to know the detail about this system image. And dig around a bit to know how to install openwrt packages via opkg command.

mimansamaheshwari, 2017-11-27:
> get_started_with_respeaker/blob/master/docs/ReSpeaker/
> QuickStart.md#respeaker-drive-unit to know the detail about this system

Thanks a lot for your valuable response but sir I am not using Respeaker
Core with Meowking drive unit I want to perform some basic voice
recognition tasks with it so still is there any need  to update Core
system? In the link you said to refer , if we use drive unit then only we
need to update system but I am not using that but still I am not able to
run the python code.

mimansamaheshwari, 2017-11-27:
Thanks a lot for your valuable response sir  but  I am not using Respeaker Core with Meowking drive unit I want to perform some basic voice recognition tasks for making a home automation system. So is there any need to update system in that case? 
After installing git is there need to install anything else to run the python code.

KillingJacky, 2017-11-27:
What Seeed product do you have? Your question is lack of information.

mimansamaheshwari, 2017-11-27:
I am using Respeaker Core.

mimansamaheshwari, 2017-11-27:
As I am using Respeaker Core only (nothing else with it) then is there any
need to install respeaker_python_library to run python code or is it
present on Respeaker core by default?