respeaker/get_started_with_respeaker #40: Alexa shows offline

jdschuitemaker, 2017-02-13:
I have added my ReSpeaker to Alexa. When checking it's status on the Settings-page in alexa.amazon.com it always is shown as offline:

[image]

I have 2 question:
1. Is there something we can do about this?
2. Is this the cause that Alexa doesn't work with TuneIn news or streaming music?

xiongyihui, 2017-02-13:
1.  It seems it is always offline when we use [alexa voice service api v1](https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/rest-overview). While, the latest alexa voice service api is [v20160207](https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/content/avs-api-overview), uses http2 and keeps a long live tcp connection. The online/offline status should be the status of http2 connection. Currently, we still use api v1, but is working on api v20160207.

2. Receiving news or streaming music is another feature which is not supported yet. Just open [an issue](https://github.com/respeaker/Alexa/issues/7) to track it. If you have alexa response message of receiving news or streaming music, just post it on [the page](https://github.com/respeaker/Alexa/issues/7), which should make us easier to implement the feature.

teknoid-, 2017-02-14:
When do you think the new API will be working?

alexeinz, 2017-02-20:
yea same question here

jdschuitemaker, 2017-03-10:
Any updates @xiongyihui ?

xiongyihui, 2017-03-13:
I think there will be an initial version at the end of this week.

jdschuitemaker, 2017-03-13:
That seems like good news to start the week with! Thank you very much!

teknoid-, 2017-03-16:
Are we still expecting an updated version of Alexa for the respeaker this week?

teknoid-, 2017-03-19:
Any update on the API V2?

xiongyihui, 2017-03-19:
Check it out at https://github.com/respeaker/Alexa/tree/v2. It's at a very early stage.

teknoid-, 2017-03-19:
When will it be ready for the respeaker?

metanav, 2017-04-09:
Any progress?

metanav, 2017-04-20:
@xiongyihui the ns/agc feature will be available to Respeaker core or specifically to mic array?

xiongyihui, 2017-04-20:
The NS and AGC will be available on Respeaker Core. The mic array already has its own NS and AGC.

sevencheng798, 2017-06-07:
@xiongyihui you can access to  https://github.com/RAKWireless/WisCore
The Link WisCore have supported Amazon Alexa function and can play music stream,e.t - such as you like song, FM iHeartRadio, iTune,Audiabl.

xiongyihui, 2017-06-07:
@sevencheng798 I didn't found any source code there. Have you found it?

closed 2017-09-06 by xiongyihui