respeaker/get_started_with_respeaker #42: Install SOX or FLAC

PRONOVATEK, 2017-02-13:
Hi, me again. Is there any method to install SOX or FLAC on respeaker, I assume that is necessary compile but I hope it is not so.

PRONOVATEK, 2017-02-13:
I try with opkg but I receive the next error:

root@ReSpeaker:~/mycroft-core# opkg install sox
Installing sox (14.4.1-3) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ramips/mt7688/packages/packages/sox_14.4.1-3_ramips_24kec.ipk.
Installing lame-lib (3.99.5-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ramips/mt7688/packages/packages/lame-lib_3.99.5-1_ramips_24kec.ipk.
Installing libvorbisidec (1.0.3-20150104-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ramips/mt7688/packages/packages/libvorbisidec_1.0.3-20150104-1_ramips_24kec.ipk.
Installing libsndfile (1.0.25-2) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ramips/mt7688/packages/packages/libsndfile_1.0.25-2_ramips_24kec.ipk.
Installing libmagic (5.25-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ramips/mt7688/packages/packages/libmagic_5.25-1_ramips_24kec.ipk.
Configuring libvorbisidec.
Configuring libmagic.
Configuring libsndfile.
Configuring lame-lib.
Collected errors:
 * check_data_file_clashes: Package sox wants to install file /usr/bin/play
	But that file is already provided by package  * respeaker-os-file
 * opkg_install_cmd: Cannot install package sox.

hope that this can contribute to answer what happens. 

regards

xiongyihui, 2017-02-14:
There is already a file named `/usr/bin/play`. You can overwrite it by running `opkg install --force-overwrite sox`.
You'd better backup `/usr/bin/play` at first, run:

`mv /usr/bin/play /root`

PRONOVATEK, 2017-02-14:
Installed!, than you for you support.

closed 2017-02-14 by PRONOVATEK