respeaker/get_started_with_respeaker #43: Cannot install software after expend storage

hengway1989, 2017-02-14:
Hi, as subject, can please help advice on this, below is the log record, thank you

root@ReSpeaker:/# df -h
Filesystem                Size      Used Available Use% Mounted on
rootfs                    1.7M      1.6M     64.0K  96% /
/dev/root                29.0M     29.0M         0 100% /rom
tmpfs                    61.7M      1.2M     60.5M   2% /tmp
tmpfs                    61.7M    136.0K     61.6M   0% /tmp/root
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/mtdblock6            5.2G     11.3M      4.9G   0% /overlay
overlayfs:/overlay        1.7M      1.6M     64.0K  96% /
/dev/mmcblk0p2            5.2G     11.3M      4.9G   0% /overlay
root@ReSpeaker:/# cd /overlay
root@ReSpeaker:/overlay# opkg install git git-http
Installing git (2.3.5-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ramips/mt7688/packages/packages/git_2.3.5-1_ramips_24kec.ipk.
Installing git-http (2.3.5-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ramips/mt7688/packages/packages/git-http_2.3.5-1_ramips_24kec.ipk.
Configuring git.
Collected errors:
 * wfopen: //usr/lib/opkg/info/git.postinst: No space left on device.
 * wfopen: //usr/lib/opkg/info/git.control: No space left on device.
 * wfopen: //usr/lib/opkg/info/git.prerm: No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-rm to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-rerere to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-diff-index to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-send-pack to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-checkout-index to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-ls-remote to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-index-pack to 'git': No space left on device.
 * wfopen: /usr/lib/git-core/git-relink: No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-ls-files to 'git': No space left on device.
 * wfopen: /usr/lib/git-core/git-merge-resolve: No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-fmt-merge-msg to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-hash-object to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-fsck to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-merge-tree to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-diff to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-update-index to 'git': No space left on device.
 * wfopen: /usr/lib/git-core/git-rebase: No space left on device.
 * wfopen: /usr/lib/git-core/git-sh-i18n: No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-log to 'git': No space left on device.
 * wfopen: /usr/lib/git-core/git-merge-octopus: No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-merge-index to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-checkout to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-prune to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-imap-send to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-name-rev to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-remote-fd to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-cherry-pick to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-interpret-trailers to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-remote to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-show-branch to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-symbolic-ref to 'git': No space left on device.
 * wfopen: /usr/lib/git-core/git-rebase--merge: No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-check-ref-format to 'git': No space left on device.
 * wfopen: /usr/lib/git-core/git-p4: No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-credential to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-grep to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-commit to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-fetch-pack to 'git': No space left on device.
 * wfopen: /usr/lib/git-core/git-pull: No space left on device.
 * wfopen: /usr/lib/git-core/git-send-email: No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-pack-redundant to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-verify-pack to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-count-objects to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-write-tree to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-archive to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-remote-ext to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-mailsplit to 'git': No space left on device.
[... 38 lines cut ...]
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-reflog to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-stage to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-clean to 'git': No space left on device.
 * wfopen: /usr/lib/git-core/git-add--interactive: No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-fsck-objects to 'git': No space left on device.
 * wfopen: /usr/lib/git-core/git-rebase--am: No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-for-each-ref to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-annotate to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-merge-subtree to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-blame to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-gc to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/bin/git-shell to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/bin/git-upload-pack to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/bin/git-upload-archive to 'git': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/bin/git-receive-pack to 'git': No space left on device.
 * set_flags_from_control: Failed to open //usr/lib/opkg/info/git.control: No such file or directory.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-remote-ftp to 'git-remote-http': No space left on device.
 * wfopen: /usr/lib/git-core/git-http-backend: No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-remote-https to 'git-remote-http': No space left on device.
 * extract_archive: Cannot create symlink from ./usr/lib/git-core/git-remote-ftps to 'git-remote-http': No space left on device.
 * pkg_write_filelist: Failed to open //usr/lib/opkg/info/git-http.list: No space left on device.
 * opkg_install_pkg: Failed to extract data files for git-http. Package debris may remain!
 * opkg_install_cmd: Cannot install package git-http.
 * opkg_conf_write_status_files: Can't open status file //usr/lib/opkg/status: No space left on device.
root@ReSpeaker:/overlay#

xiongyihui, 2017-02-16:
It seems the sd card doesn't work properly. The rootfs has very limited space left.

>rootfs 1.7M 1.6M 64.0K 96% /

Maybe, try to format the sd card using [SD Card Formatter](https://www.sdcard.org/downloads/formatter_4/) and try again.

shannahs, 2017-08-24:
由于rootfs 1.7M 1.6M 64.0K 96% /     ，重新格式化sd卡后仍无法修改fstab，有哪些文件可以删除不？释放掉一些空间后应该就能操作了

xiongyihui, 2017-08-24:
可以运行 `first_boot` 恢复出厂设置，然后把sd卡用作extroot

shannahs, 2017-08-24:
/etc/init.d/mopidy stop指令正常执行
/etc/init.d/alexa stop执行出错导致python playmusic.py
运行时老是出现Detected alexa。。怎么解决

xiongyihui, 2017-08-25:
运行 `python playmusic.py` 出现 Detected alexa，这是错误的语音识别导致。
可以修改 `/usr/lib/python2.7site-packages/respeaker/pocketsphinx-data/keyword.txt`，增长alexa的检测阈值。比如：

```
respeaker /1e-30/
alexa /1e-20/
play music /1e-40/
```

shannahs, 2017-08-25:
想问问，播放音乐的格式支持几种，能不能支持无损音频flac和wma

xiongyihui, 2017-08-25:
gstreamer已经内置了，也可以安装ffmpeg，大部分音乐格式都可以支持吧，flac是可以的，wma没测过

KillingJacky, 2017-10-17:
将关闭此issue, 如有其他问题请新建issue😐

closed 2017-10-17 by KillingJacky