respeaker/get_started_with_respeaker #44: Noise on mic array

bwhitman, 2017-02-14:
Asked this over on the forum but i guess we are supposed to ask here:

I am getting inevitable loud corrupted frames from the mic array while connected over USB after a certain amount of time. It varies. It will sound great for a while and then eventually start slowly corrupting the audio until you can't hear it, even after successive recording attempts. It feels like something in the XMOS firmware is not working properly. Here's an example of the noise I'm getting:

https://soundcloud.com/bwhitman/respeaker-mic-array-test

I am just saying "test" alone in a quiet room; all the other noises seem artificial.

elpimous, 2017-02-14:
Yes.
I ´m impatient to hear solution...
Some of my friends and i wait for solution
Vincent

Fuhua-Chen, 2017-02-17:
@elpimous @bwhitman ，
I can not find the track by your link.
It seems caused by AGC. Once power up the mic array, the AGC run with a certain gain, and because you are in a quiet room, the gain will be adjusted to a larger value.
please have a try to close the AGC and set a low gain to find that it is the root cause or not by mic array HID tool.
[HID-tool](https://github.com/Fuhua-Chen/ReSpeaker-Microphone-Array-HID-tool)

bwhitman, 2017-03-09:
Even with AGC off (setting 0x2A to 0, and reading it back to confirm) I still get noise in the middle of the recording (it gets worse over time.)

gdpeter8075, 2017-12-19:
I set 0x2A and 0x2B to zero. I can tell it does noise suppression. But the voice recording quality is not good if I stand more than 5m away from the mic array. My noise source comes from youtube playing white noise and it is 1 meter from my mic array. What register value should I change?? thank you