respeaker/get_started_with_respeaker #47: Factory reset doesn't work

mineub, 2017-02-15:
hi,

i have tried fristboot and the other way copieing the script on SD Card for autorun which was explained in https://github.com/respeaker/get_started_with_respeaker/wiki/factory-reset, but both ways doesn't work. at firstboot i get error back "/dev/mtdblock6 is not mounted
/dev/mtdblock6 will be erased on next mount". sd autorun doesnt do anything. If i remove SD card and try to start without but respeaker wont work without sdcard. nothing is working. at df -h following is printed:

Filesystem                Size      Used Available Use% Mounted on
rootfs                    5.2G     19.1M      4.9G   0% /
/dev/root                29.0M     29.0M         0 100% /rom
tmpfs                    61.7M    284.0K     61.5M   0% /tmp
/dev/mmcblk0p2            5.2G     19.1M      4.9G   0% /overlay
overlayfs:/overlay        5.2G     19.1M      4.9G   0% /
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/mmcblk0p1            2.0G     40.6M      2.0G   2% /mnt/mmcblk0p1

i dont know what to do anymore. at the moment respeaker is a very expencive paperweight. Nothing is working :-(

PLEASE help me, 
michael

xiongyihui, 2017-02-16:
Factory reset only resets the contents of the flash (/dev/mtdblock2) and won't change the sd card. If you want to clear the content on sd card. Try to use SD card formatter to format the sd card.

pedronf65, 2017-02-22:
I can't connect to Respeaker WiFi any more so I want to make a factory reset, but I'm having the same issue:
`root@ReSpeaker:/# firstboot`
`This will erase all settings and remove any installed packages. Are you sure? [N/y]`
`y`
`/dev/mtdblock6 is not mounted`
`/dev/mtdblock6 will be erased on next mount`
How can I do a factory reset and start over?
Thanks in advance,
Pedro

xiongyihui, 2017-02-22:
Did you remove your sd card?

pedronf65, 2017-02-22:
No... do I need to?

pedronf65, 2017-02-22:
Ok, I removed the SD card and now the factory reset worked! Thanks xiongyihui

closed 2017-02-28 by jerryyip

wuchouk, 2017-04-18:
Hi,

Even after I removed SD card, it still shows
/dev/mtdblock6 is not mounted
/dev/mtdblock6 will be erased on next mount
what should I do?

Thanks

jerryyip, 2017-04-26:
@wuchouk That is fine. Just reboot and respeaker will be back to factory.

usernamepasswordBS, 2025-04-15:
But it's not though. df -h still shows overlay which is not factory I believe.