respeaker/get_started_with_respeaker #48: Web Terminal is stake I can't do any thing

zezokhaled84, 2017-02-15:
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the RSA key sent by the remote host is
SHA256:kz1Y2yDY9ETD61QGtzLsHkfpmgxe/CQl+ONvPunCV5g.
Please contact your system administrator.
Add correct host key in /root/.ssh/known_hosts to get rid of this message.
Offending RSA key in /root/.ssh/known_hosts:1
RSA host key for localhost has changed and you have requested strict checking.
Host key verification failed.

zezokhaled84, 2017-02-15:
Logs
Wed Feb 15 16:31:53 2017	auth.err	sshd[2135]	error: Could not load host key: /etc/ssh/ssh_host_ecdsa_key
Wed Feb 15 16:31:53 2017	auth.err	sshd[2135]	error: Could not load host key: /etc/ssh/ssh_host_ed25519_key

xiongyihui, 2017-02-16:
Can you access the system through ssh using ssh client like putty?

zezokhaled84, 2017-02-16:
Yes I could 
But the web terminal give this error

xiongyihui, 2017-02-28:
Did you run `firstboot` to reset the system with a sd card?  Maybe remove `/root/.ssh` and try again.

yyd19948, 2017-07-14:
你最后怎么解决的啊