respeaker/get_started_with_respeaker #50: how to change wake up word?

calvinwongsc, 2017-02-17:
does anyone have change the wake up word ?

calvinwongsc, 2017-02-20:
i have changed 2File txt file in root/respeak_python_library/pocketsphinx-data/
but i still can't edit changed the wake up word...
looking for someone to help~~~

aleifuu, 2017-02-21:
I initially thought it would be as simple as changing the word inside the wakeup function ... like

`if mic.wakeup('respeaker'):`

but apparently not ... any response dev/ team ? :D

xiongyihui, 2017-02-28:
We need to add new words to `pocketsphinx-data/keyword.txt` and `pocketsphinx-data/dictionary.txt`, and then change the code
```
if mic.wakeup('respeaker'):
```

alexeinz, 2017-03-13:
I have same problem.. changed keyword in alexa and defined it in both files but alexa.py just crashing :-1: 
root@ReSpeaker:~# alexa                                                                                                                                         ♦☺@                                                                                                                                                             ALSA lib pcm_dmix.c:961:(snd_pcm_dmix_open) The dmix plugin supports only playback stream                                                                       ALSA lib pcm_dsnoop.c:557:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream                                                                  ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave                                                                                               INFO:mic:Use ReSpeaker MicArray UAC2.0: USB Audio (hw:1,0)                                                                                                      Traceback (most recent call last):                                                                                                                                File "/usr/bin/alexa", line 368, in <module>                                                                                                                      main()                                                                                                                                                        File "/usr/bin/alexa", line 341, in main                                                                                                                          mic = Microphone(quit_event=quit_event)                                                                                                                       File "/usr/lib/python2.7/site-packages/respeaker/microphone.py", line 97, in __init__                                                                             self.decoder = self.create_decoder()                                                                                                                          File "/usr/lib/python2.7/site-packages/respeaker/microphone.py", line 129, in create_decoder                                                                      return Decoder(config)                                                                                                                                        File "/usr/lib/python2.7/site-packages/pocketsphinx/pocketsphinx.py", line 326, in __init__                                                                       this = _pocketsphinx.new_Decoder(*args)                                                                                                                     RuntimeError: new_Decoder returned -1

xiongyihui, 2017-03-13:
@alexeinz Did you add any keyword to the `keywords.txt` in which empty line is not permitted ?
There is a note to add new keyword - https://github.com/respeaker/get_started_with_respeaker/issues/68#issuecomment-285596068

alexeinz, 2017-03-13:
yes I added both to keywords.txt and dictionary.txt and i dont have new line in both.. just doublechecked...

jacksun101, 2017-04-01:
I've followed each steps @xiongyihui said, but it just doesn't recognize the new keyword. Any idea?

xiongyihui, 2017-04-01:
Which new keyword did you add? If the keyword is less common, it will be hard to recognize.

Here is an example to add keywords `ok` and `stop`:

1. keywords.txt
```
respeaker /1e-30/
alexa /1e-30/
play music /1e-40/
ok /1e-45/
stop /1e-45/
```

2. dictionary.txt
```
respeaker	R IY S P IY K ER
alexa	AH L EH K S AH
play	P L EY
music	M Y UW Z IH K
ok	OW K EY
play	P L EY
stop	S T AA P
```

jacksun101, 2017-04-01:
@xiongyihui Where are your txt files located? For me, files are under the following path.

/root/respeaker_python_library/respeaker/pocketsphinx-data

And even if I remove the line of "alexa", it can still recognize it. So, I am wondering...
BTW, my respeaker_python_library is cloned from the latest repo of github.

jacksun101, 2017-04-01:
OK, I got it. Correct file path for me is as follows. 

/usr/lib/python2.7/site-packages/respeaker-0.6.0-py2.7.egg/respeaker/pocketsphinx-data

sudhaa1232, 2017-04-21:
have you done this?

jacksun101, 2017-04-21:
Yes, for me it works. The point is to edit the correct files, and they depend on the python script you're running.

closed 2017-10-17 by KillingJacky