respeaker/get_started_with_respeaker #51: Converting ReSpeaker documentation to MkDocs

d34dman, 2017-02-17:
http://www.mkdocs.org/ Provides excellent support for documentation written in markdown.

I have forked this repo and made it compatible with MkDocs already, Check the final result here https://d34dman.github.io/get_started_with_respeaker/.

Advantages
1. Better browsing experience
2. Supports search
3. Provides Edit link on the document itself, while it is being viewed.
4. Cleaner navigation architecture (Autodiscovery of headings)
5. We get support for admonitions which are cool features that can be used to attract user attention. (for example special theming for "note", "alert", "warning" etc...)

The Catch

1. MkDocs doesn't fully support GitHub flavoured markdown. So far the only incompatibility i have noticed is the use of "```" along with "4 spaces" intend to highlight code. The workaround is simple. We have to stop using "4 spaces" intend.

2. To publish the document, we would have to perform an extra step. Basically `mkdocs gh-deploy` has to be run in the root directory to publish the document on GitHub Pages. Alternatively if we host the document on readthedocs.io then we don't have to worry about this. They implement a githook which automatically builds the document upon change.
(Readthedocs integration can be tested at http://get-started-with-respeaker.readthedocs.io/en/latest/)

--

Let me know if you like the concept and i can help you port your latest document HEAD over to MKDocs.

---

d34dman, 2017-02-17:
I have send an invitation to @xiongyihui so that he can check how "Edit on GitHub" link works :)

xiongyihui, 2017-02-17:
Great job! Thank you, @d34dman

d34dman, 2017-02-17:
Glad you liked it, evaluate and let me know if you would like to use it as official documentation. I can fix the code highlighting issue in certain area and we would be good to go.

jerryyip, 2017-02-28:
Hi, @d34dman. Thanks for your great job! But that is a shame that we don't have a plan to convert our wiki to mkdocs. Thanks again! And the issue will be closed.

closed 2017-02-28 by jerryyip