respeaker/get_started_with_respeaker #52: issue running bing_speech_api.py

quezbeme, 2017-02-18:
So when I run the bing_speech_api.py I get this error:
INFO:mic:etected respeaker
000000000000000000000000000000000000000000000000000Wake up
INFO:mic:Start listening
got data
DEBUG:requests.packages.urllib3.connectionpool:"POST /token/issueToken HTTP/1.1" 401 143
I think this is an auth issue so I'm wondering where should the bing key go besides playmusic.py and if there's something else I'm missing. Thanks!

quibusus, 2017-02-18:
looks like you're running old version of respeaker python API, looks like the latest (0.6) is not published properly

- connect to your respeaker (see get started, don't forget to install git)
- build latest version like this
`
git clone https://github.com/respeaker/respeaker_python_library.git
cd respeaker_python_library
python setup.py install
`

quezbeme, 2017-02-19:
I had already done that but I redid it just in case. Now I get this:
No.0 try
TTS:Your beliefs become your thoughts
INFO:requests.packages.urllib3.connectionpool:Starting new HTTPS connection (1): api.cognitive.microsoft.com
DEBUG:requests.packages.urllib3.connectionpool:"POST /sts/v1.0/issueToken HTTP/1.1" 401 143
Could not request results from Microsoft Bing Voice Recognition service; http request error with status code 401
7.72485494614
No.1 try
TTS:Your thoughts become your words
DEBUG:requests.packages.urllib3.connectionpool:"POST /sts/v1.0/issueToken HTTP/1.1" 401 143
Could not request results from Microsoft Bing Voice Recognition service; http request error with status code 401
0.362385034561
No.2 try
TTS:Your words become your actions
DEBUG:requests.packages.urllib3.connectionpool:"POST /sts/v1.0/issueToken HTTP/1.1" 401 143
Could not request results from Microsoft Bing Voice Recognition service; http request error with status code 401
0.0745718479156
No.3 try
TTS:Your actions become your habits
DEBUG:requests.packages.urllib3.connectionpool:"POST /sts/v1.0/issueToken HTTP/1.1" 401 143
Could not request results from Microsoft Bing Voice Recognition service; http request error with status code 401
0.0723371505737
No.4 try
TTS:Your habits become your values
DEBUG:requests.packages.urllib3.connectionpool:"POST /sts/v1.0/issueToken HTTP/1.1" 401 143
Could not request results from Microsoft Bing Voice Recognition service; http request error with status code 401
0.07843708992
No.5 try
TTS:Your values become your destiny
DEBUG:requests.packages.urllib3.connectionpool:"POST /sts/v1.0/issueToken HTTP/1.1" 401 143
Could not request results from Microsoft Bing Voice Recognition service; http request error with status code 401
0.0933880805969

quibusus, 2017-02-19:
Status code 401 means unauthorized, check your api key

quezbeme, 2017-02-20:
That's what I figured and I've used both and regenerated both keys multiple times. Could there be any reason the keys wouldn't work?

quibusus, 2017-02-21:
Try this curl, if it works the problem is somewhere in the respeaker lib. 

`
curl -v -X POST "https://api.cognitive.microsoft.com/sts/v1.0/issueToken" -H "Content-type: application/x-www-form-urlencoded" -H "Content-Length: 0" -H "Ocp-Apim-Subscription-Key: your_subscription_key"
`
(found here)

https://www.microsoft.com/cognitive-services/en-us/Speech-api/documentation/GetStarted/GetStarted-cURL#Step1

mkeymont, 2017-09-01:
I ran into the same error and the problem turned out to be that my respeaker_pythin_library folder was empty, and it was a new device, so I hadn't done anything to it to cause it to be empty. I removed the directory using rm -rf and then cloned the library, installed it and all was well.

Also, make sure mo-idy is stopped or you will have trouble as well.

mkeymont, 2017-09-05:
What I can't figure out, though is, if I have to stop Mopidy to be able to use the microphone for recognition, what good is Mopidy at all? My goal is to be able to call up Spotify playlists via voice command. That doesn't seem to be possible with ReSpeaker due to this limitation.

Am I missing something?