respeaker/get_started_with_respeaker #53: airplay issue while alexa is running

alexeinz, 2017-02-18:
While Alexa.py is running, even though respeaker is still active as Airplay device , the sound doesn't go thru or does with very low quality / bitrate.

quezbeme, 2017-02-19:
I have this same issue. In order for me to stream to the respeaker I have to disable mopidy (and then restart it when I'm done using it as an Airplay device). Unfortunately when I stop mopidy, any web based apps stop working, like web term, music player, or file manager. Dashboard still works.

alexeinz, 2017-02-19:
I dont think its related to mopidy specifically, airplay works fine with mopidy disabled and stopped.
if airplay active before i got the following when running alexa.py
DEBUG:./alexa.py:Play DeviceTTSRenderer_466de1f3-b9bf-4275-9723-ebb6b262f0fc_461940417
aplay: main:722: audio open error: Device or resource busy

alexeinz, 2017-02-19:
yep, basically if anything is playing thru , like airplay audio , alexa listens but cant reply over the audio
INFO:mic:Start detecting
INFO:mic:Detected alexa 
DEBUG:root:wakeup

INFO:mic:Start listening
DEBUG:./alexa.py:Start sending speech to Alexa Voice Service
1111+11111111111100000000-000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
INFO:mic:Stop listening
DEBUG:./alexa.py:Finished sending speech to Alexa Voice Service
DEBUG:./alexa.py:Processing Request Response...
DEBUG:./alexa.py:JSON String Returned: {
  "messageBody": {
    "directives": [
      {
        "namespace": "Speaker", 
        "name": "SetMute", 
        "payload": {
          "mute": false
        }
      }, 
      {
        "namespace": "SpeechSynthesizer", 
        "name": "speak", 
        "payload": {
          "contentIdentifier": "amzn1.as-ct.v1.Domain:Application:Notifications#ACRI#DeviceTTSRenderer_c07255e1-203a-4c87-b563-feddafed8d57", 
          "audioContent": "cid:DeviceTTSRenderer_c07255e1-203a-4c87-b563-feddafed8d57_1869891702"
        }
      }
    ]
  }, 
  "messageHeader": {}
}
DEBUG:./alexa.py:Play DeviceTTSRenderer_c07255e1-203a-4c87-b563-feddafed8d57_1869891702
MPEG Audio Decoder 0.15.2 (beta) - Copyright (C) 2000-2004 Robert Leslie et al.
aplay: main:722: audio open error: Device or resource busy
41 frames decoded (0:00:00.9), -3.0 dB peak amplitude, 0 clipped samples
DEBUG:./alexa.py:SpeechSynthesizer audio: DeviceTTSRenderer_c07255e1-203a-4c87-b563-feddafed8d57_1869891702

alexeinz, 2017-02-19:
would there be a way to detect speaker use, and to pause or overlay using the alsa device ?

aleifuu, 2017-02-20:
@alexeinz, are you using MicArray extension btw ?

I just received respeaker kit yesterday and started to play around with it. I have attached MicArray to broaden its 'hearing' area up to 4-5 meters, but like you, whenever executing the `alexa` program. It seems greatly reduced its 'hearing' capability ... the led would light up indicating some words, but failed to catch the wake up word alexa itself from a distance. I will have to get close like 1m from respeaker in order for it to catch the word alexa correctly

quezbeme, 2017-02-20:
I see what you mean. I installed Alexa and have to stop running it in order to use airplay.

alexeinz, 2017-02-20:
Yea, the mic array definitely makes things worse... I just removed mine and it seem to work much better ... seems useless waste of money... can anyone from dev pitch in here?

xiongyihui, 2017-02-28:
We overestimate the performance of the MT7688. To use airplay (Shairport-sync), we should stop alexa.
We'd better make it do one thing and do it well.

Do you think we should provide different firmware, for example , one for alexa voice service, one for airplay, mopidy?

alexeinz, 2017-02-28:
do you think so? Alexa actually works fine with airplay here, and the problem is not in performance but more in that the applications take the channels . 
why not use something like Snapcast to be able to route/share channels for  the multiple audio https://github.com/badaix/snapcast/blob/master/doc/player_setup.md#airplay .

Typically an airplay house - would use airplay and upnp household would use mopidy , wouldnt it make more sense to do either tuned airplay or mopidy as the choices for diff firmware..

xiongyihui, 2017-02-28:
When alexa is running, we can still use `aplay` to play sound. When Alexa is not playing sound, other application can use the output channel to play music.

closed 2017-10-17 by KillingJacky