respeaker/get_started_with_respeaker #54: tmpfs too small - preventing software installation

alexeinz, 2017-02-18:
tmpfs too small - preventing software installation on some larger packages.
is there a reason its limited to 60mb?

jerryyip, 2017-02-21:
We used [MT7688](https://labs.mediatek.com/en/chipset/MT7688) SoC.
BTW, you could use a SD card to [extend storage](https://github.com/respeaker/get_started_with_respeaker/blob/master/QuickStart.md#use-sd-card-to-extend-storage) easily.

alexeinz, 2017-02-21:
I am talking about tmpfs which seems limited by default to 64mb... i wonder if there is a reason for it, its definitely not enough when pooling large packages with wget

alexeinz, 2017-02-22:
@jerryyip why not use max ram then ?
Supports up to 256MB of external 16-bit DDR1/ DDR2 (193MHz) memory

xiongyihui, 2017-02-28:
We need to give some parts of the RAM to the linux kernel and applications.

To download large file, how about download it into SD card instead of tmpfs?

closed 2017-10-17 by KillingJacky