respeaker/get_started_with_respeaker #56: squeezebox client

alexeinz, 2017-02-20:
do you think it would be possible to integrate Squeezebox client ? 
http://forums.slimdevices.com/showthread.php?104022-OpenWrt-Squeezelite-Package-Released

xiongyihui, 2017-02-28:
Did you try to compile the package and install the ipk into the respeaker?

alexeinz, 2017-02-28:
should be done by end of the week, didnt have time to wrk on it
here are the steps if you have the time and all set up https://github.com/squeezewrt/squeezewrt
btw.. any reason you using that weird Juci angular thingie and not standard Lucy

xiongyihui, 2017-02-28:
There are music player and file manager using Angular too, so we think it should be easier for developers to work with them. It seems Juci is not mature enough.

alexeinz, 2017-03-09:
Squeezelite is compiled and working great ,I can play no problem
you do need change options in /etc/config/squeezelite as follows

config options 'options'
        option name 'SqueezeWRTLivingroom'
        option model_name 'SqueezeLite'
        option close_delay '0'
        option specific_server '0'
        option priority '0'
        option specific_bufsiz '0'
        option max_sr '0'
        option specific_devopen '0'
        option device 'hw:0,0'
        option decode_flac '1'
        option decode_mp3 '1'
        option decode_aac '1'
        option decode_ogg '1'
        option decode_wma_alac '1'
        option dsd_over_pcm '0'
        option ircontrol '0'
        
and
in /etc/init.d/squeezelite 
remove the -i option or it wont start....

[squeezelite_1.8-1_ramips_24kec.ipk.zip](https://github.com/respeaker/get_started_with_respeaker/files/832535/squeezelite_1.8-1_ramips_24kec.ipk.zip)

alexeinz, 2017-03-10:
@xiongyihui  btw see above.. you can add the squeezebox to ur wiki if u want

closed 2017-03-10 by alexeinz