respeaker/get_started_with_respeaker #57: respeaker mic array issues

alexeinz, 2017-02-21:
Respeaker mic array when added causes significant slow down on the device.
Voice recognition also not improved atleast with Alexa.

aleifuu, 2017-02-21:
+1 on this issue ... unless people from dev/ could give us how-to fine tune it

What I experience is that MicArray seems vastly increase the distance of the hearing area for ReSpeaker, but also vastly dumbs down catching up the keyword ... 'alexa' in this example

However I only find difficulties of ReSpeaker catching the keyword. After it wakes up, I think it somehow receives better even from a distance ( well, I need to shout it a little ) 

I don't think it's the intended use of this MicArray ( to shout to it ) ... people from dev/ ?

toxin20, 2017-02-25:
I have the same problem. The Keyword detection ("alexa" in my case) is very unreliable even though I have the MicArray.

Eternyt, 2017-02-26:
Same here. Both "respeaker" and "alexa" are very few times recognize correctly(with the mic array). Also, if in the room there are some noise (like two people that are speaking) alexa wakes up randomly without any recognizable word "alexa" pronounced.

alexeinz, 2017-02-28:
Hello devs? the mic array is not working... anyone wants to clarify ?

Fuhua-Chen, 2017-02-28:
It seems that the software DSP core inside xmos controler behaves differently in different enviroment. We will update the firmware ASAP once XMOS updates it.
Also, we are going to release a firmware exclude DSP core. It will output 7 channels raw audio data via usb.

aleifuu, 2017-03-02:
Hi @Fuhua-Chen or dev/ in general, 

Relating to previous posts about the firmware update ... would the update likely this problem or do we still need to increase the threshold in `keyword.txt` ? which way will work better to address the problem that we are having with the mic array ?

shinselrobots, 2017-03-21:
Is there an update on this issue?
What firmware version had the problem?  Is it fixed or still broken in v3.1?
"we are going to release a firmware exclude DSP core"  This would help as a debug tool, but does not solve the real problem (we want a far field microphone that works)!

Great Potential for this product, but needs better support!!!
Can someone write a clear "getting started with stand-alone microphone array" document?

elpimous, 2017-03-22:
yes i'd like a tuto, too, from install, root access, and use for microphone array on ubuntu 
please.
best regards
vincent

shinselrobots, 2017-03-22:
P.S.  If you fix these issues, I will order 15 more immediately!  We need these for an internal project.

cecio, 2018-01-04:
Hi @Fuhua-Chen,
do you have any news on this issue? 
Thanks

scher200, 2018-03-31:
I heared about the improved version! When will this new version reach the market?

elpimous, 2018-03-31:
Hi. It is written april !

scher200, 2018-03-31:
awesome

elpimous, 2018-03-31:
Well... i need it too !
But Seeed don't have real delay...

I think we'll have to wait more !