respeaker/get_started_with_respeaker #58: I can't connect to the respeaker via SSH and the blue lamb is flashing

zezokhaled84, 2017-02-21:
I can't connect to the respeaker via SSH and the blue lamb is flashing
I need help please as fast as you can

aleifuu, 2017-02-22:
have you tried to reboot it ? if not try accessing it from the Web Terminal

zezokhaled84, 2017-02-22:
No webterminal no serial no wifi

aleifuu, 2017-02-22:
seems most unfortunate ... and the thread has been quite empty no response from dev/ either ... oh well

just out of curiosity did you experience sudden power outtage or something ? that may 'fry' the power on respeaker ? how big the charge you put onto respeaker ?

zezokhaled84, 2017-02-22:
Nothing special 
Its power on normal even now
The green lamb is power and the blue is flashing

The power only the mac usb power

jerryyip, 2017-02-22:
could you find /dev/tty.usbmodem**** in your mac? That is respeaker.
"screen /dev/tty.usbmodem****" to login via serial

zezokhaled84, 2017-02-22:
How can it find this?

I try from the terminal to connect ssh to respeaker and the host showing down

aleifuu, 2017-02-22:
I think what @jerryyip meant is that you don't ssh to it but rather you screen to it. I don't own a Mac, but you should be able to execute it from terminal

jerryyip, 2017-02-22:
@zezokhaled84  You could screen to it and check what is the problem in the terminal.

zezokhaled84, 2017-02-22:
Can you explain to me what is the meaning of screen to it? I have low experiance

jerryyip, 2017-02-23:
[image]

This starts serial communication with your respeaker, like ssh.

aleifuu, 2017-02-25:
how's the progress ? can you screen to it ?

zezokhaled84, 2017-02-25:
I managed to recover via the aurdrino scratch

jerryyip, 2017-02-28:
It is a serial port monitor on Mac, just like putty on Windows. @aleifuu

jerryyip, 2017-02-28:
How is it going? @zezokhaled84

Ervanu, 2017-05-09:
I have the same problem with my ReSpeaker core v1.0 10/13/2016. When i connect it with a PC/Laptop via USB, Com Port is undetectable and wifi is also undetectable. Green light on, blue light Flashing.

What is the solution, Please? :(

michaelsit91, 2017-08-19:
I'm having the same problem now, no wifi no com port, arduino does not detect the board, any suggestions?

closed 2017-11-27 by KillingJacky