respeaker/get_started_with_respeaker #59: Respeaker too Hot

notnatech, 2017-02-21:
Hi,

I left my Respeaker running over night, the following day, I can't type to the terminal. It's very slow. It took about 40 seconds to display the last key that I pressed. 

I checked the unit and touch to see how got it is. It was probably in the 60C .  

Anyone encountered the same issue?

aleifuu, 2017-02-22:
I haven't let my ReSpeaker run over night ... so no have not encountered that issue yet

Anyhow, is it still working now ? and do you install micArray on top of the core ?

notnatech, 2017-02-22:
Yep still working, but isnt that what its supposed to do? Leave it running 24x7?

aleifuu, 2017-02-22:
I just got this device about 2 days and was intending to run it 24 hours x 7 provided it has been setup correctly 

I run into problem whenever MicArray inserted, it does seem widen its capability to detect sound from afar but at the same time introduces more noise ... having hard time to detect the keyword like alexa from respeaker ... unless you shout

Do you not run into the same issue btw ?

alexeinz, 2017-02-22:
I think thats the mic shield, my respeakers functions much better after removing it... the devs seem totally deserted this project though... ,since its actually an hid interface, maybe we could just use usb speaker / mic connection on pinout ?

aleifuu, 2017-02-22:
Hey @alexeinz, just out of curiousity since you seem to be the more active user here. Does your stop working with alexa services at the moment ? cause mine does. It was alright yesterday

Now the led just turns red and stuck there ... usually it would go red-dissapear to ready listening state. but this one just stuck turning red

alexeinz, 2017-02-22:
just remove mic array and re-initiated the whole thing today with Alexa running as daemon... wanna compile squeezebox for it so gonna leave it running few days and see.

notnatech, 2017-02-22:
I am not using Alexa, I think it's too bloated. I am using the core respeaker python code and customize it based on my need. 

My Mic Far Field seems to work fine, in fact, I experience the opposite whenever I unplug the Mic Array from the core.

aleifuu, 2017-02-22:
do you still encounter the overheating issue, @notnatech ?

I'm wondering do you power up the mic far field separately or by the bridge between the core and mic array ?

core respeaker python code ... I don't think you are using Bing either. Which files are you based on ? I'd like to customize the voice command based on my need too

notnatech, 2017-02-24:
I am powering up via respeaker core. I am using Bing, here is my setup:

https://www.youtube.com/watch?v=PXUCZYE492c&feature=youtu.be

aleifuu, 2017-02-25:
nice setup @notnatech !

I try leaving it running overnight. It's charged via samsung power adapter like how you would your phone. It doesn't get too hot, lukewarm ish ... not enough to raise for alarm. I'm just using my ReSpeaker core and not /w the array. is yours using /w the array ?

aleifuu, 2017-02-25:
@notnatech , also btw how do you change the wake up word ? I try change it to something like ... 'butler' ... but the Mic only detects either 'alexa' or 'respeaker' 

The code I changed is this one `mic.wakeup('butler')`

notnatech, 2017-02-25:
Hi @aleifuu, I am using the respeaker_hi code and not the pre-built speech recognition packages. 

https://github.com/respeaker/respeaker_hi

You need to modify the model/keyword.txt and model/respeaker.dic and add your custom keywords.

aleifuu, 2017-02-26:
Thanks @notnatech , looks like the right files I finally can start my own project with ReSpeaker

notnatech, 2017-02-27:
Developers??? Anyone??? What's the use of this github project if no developer is looking into it?

notnatech, 2017-03-07:
Wow.... More than 10 days and still no response from the developers. Kickstarter page told us to post questions here directly to developers, but it looks like no response from here either... Poor customer support and maybe lack of technical skills as well?

jerryyip, 2017-03-10:
Hi @notnatech,  60C is normal for MT7688.

closed 2017-10-17 by KillingJacky