respeaker/get_started_with_respeaker #60: speaker breakout - separating output

alexeinz, 2017-02-22:
Apparently , both mic array and the respeaker itself have aux out 
How can I separate output of lets say alexa responses to go to internal respeaker out ,vs using the 3.5 output ? 
How can we use the ( non soldered for some weird reason) output on the mic array module ? 
programmatically with examples please.

alexeinz, 2017-03-09:
so apparently if i solder the mic array output out and set it as output that can be used...!

xiongyihui, 2017-03-10:
When using the mic array with the respeaker core, The aux out of the respeaker core can be the output of the mic array or the respeaker core, so the audio jack of mic array is not solderd, it's for these who want to use it with raspberry pi, pc or mac.
Besides, the respeaker core has the preemption of the aux out. It means if app1 plays sound using mic array and app2 plays sound using on-board codecs, we will hear the sound of the app2.

alexeinz, 2017-03-10:
ok, thats maybe better, so what you mean is instead of how now applications running are fighting for output , as long as i for example , direct all alexa to i2s and the airplay to mic array, sound will just pause? 
can i disable the premption btw and totally separate the outputs?

closed 2017-10-17 by KillingJacky