respeaker/get_started_with_respeaker #61: How do I use Alexa on ReSpeaker with my own credentials ?

aleifuu, 2017-02-23:
Hi dev/

The current step to use Alexa on ReSpeaker is to use your executables with your credentials. How do I use my own product_id, client_id, client_secret ? so that I can fully customize Alexa experience on my own term ?

Many thanks,

alexeinz, 2017-02-24:
edit /usr/bin/alexa in vi

aleifuu, 2017-02-25:
@alexeinz , I did edit the alexa executables to insert my own client_id and client_secret but running it make it output error `Segmentation fault`

Curious maybe it's got something to do with the web_auth process that I have to redo/clear because I was using ReSpeaker's credential beforehand ...

alexeinz, 2017-02-28:
hmm, i had no issue, what i did was
1.edit the alexa file in /usr/bin and chmod it again for anycase
2.created blank alexa.json file in /etc/config

whitelisted the ips in amazon dev for that device ( internal ip of the respeaker ( the interface connected to wan ) )
and reauthorized.
works fine

aleifuu, 2017-02-28:
@alexeinz okay, let me try that later on. Thanks !

aleifuu, 2017-03-12:
I think it's working. I just need to whitewash the ip and I think that's it. 

Hey you know what, how do you use / incorporate Alexa in ReSpeaker for ? I wonder how people are actually using it the service now that they have ReSpeaker on hand

alexeinz, 2017-03-13:
using it with openhab to ctrl my house.. but havent had much success yet tbh... both of my respeakers with mic array or not never stay live more than a day... i am troubleshooting but starting to get frustrated