respeaker/get_started_with_respeaker #62: poweroff command

notnatech, 2017-02-23:
Hi,

poweroff command doesn't really shutdown the unit. Is there any other special command to turn the unit powered off?

aleifuu, 2017-02-25:
hmm, I'm glad you raised this issue. I thought that's what it does ( or supposedly ) ... would love to know if there are proper way to shutdown it

xiongyihui, 2017-02-28:
Neither the mt7688 has a poweroff mode, nor the openwrt system has a poweroff command, so `poweroff` is removing the power supply.

aleifuu, 2017-02-28:
hi @xiongyihui ,  so do we : 

a) still issue` poweroff `command before we plug out the power source ?
or
b) just plug the power source ...

xiongyihui, 2017-02-28:
It doesn't have a `poweroff` command. So just plug the power source.

aleifuu, 2017-03-02:
I think this one is pretty much answered. Can you close it ? so it's not dangling around

closed 2017-03-02 by notnatech