respeaker/get_started_with_respeaker #63: amazon login fail ?

xiaooquanwu, 2017-02-24:
“At the first time, you need to authorize the application.

Connect ReSpeaker's Access Point, go to http://192.168.100.1:3000 and tt will redirect you to Amazon to sign up or login in.”

do anyone success for that step ?

alexeinz, 2017-02-24:
see this 
https://github.com/respeaker/Alexa/issues/8#issuecomment-281939757
basically, the documentation is wrong for the ipk 
1.to get authenticated, you need to create alexa device and whitelist the internal IP for your device
2.the client and secret needs to be replaced in the /usr/bin/alexa because the data there is a dev left over info.... 
3.I suggest to create alexa.json in /etc/config because it doesnt create it by itself ( just blank file )

xiaooquanwu, 2017-02-24:
you need to create alexa device and whitelist the internal IP for your device ?

Do not Understand ，Can you tell me how to do ?

alexeinz, 2017-02-24:
https://github.com/respeaker/Alexa
https://sudodblog.wordpress.com/2016/04/05/lets-make-an-amazon-alexa-device/

xiaooquanwu, 2017-02-24:
For ReSpeaker ,the Web Setting On avs is that:
Allowed Origins: http://192.168.100.1:3000
Allowed Return URLs: http://192.168.100.1:3000/authresponse
right？but still fail

alexeinz, 2017-02-24:
whats the ip of your respeaker? 
if you joined it to your local network it will have dhcp assigned ip
whitelist that

xiaooquanwu, 2017-02-24:
root@ReSpeaker:/etc/config# ifconfig 
apcli0    Link encap:Ethernet  HWaddr 9E:65:F9:0E:00:7A  
          inet addr:10.1.1.218  Bcast:10.1.1.255  Mask:255.255.255.0
          inet6 addr: fe80::9c65:f9ff:fe0e:7a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

br-lan    Link encap:Ethernet  HWaddr 9C:65:F9:1D:F4:BE  
          inet addr:192.168.100.1  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:61474 errors:0 dropped:0 overruns:0 frame:0
          TX packets:191101 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:6583404 (6.2 MiB)  TX bytes:101829878 (97.1 MiB)

eth0      Link encap:Ethernet  HWaddr 9C:65:F9:1D:F4:BE  
          inet6 addr: fe80::9e65:f9ff:fe1d:f4be/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:12513 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:1442816 (1.3 MiB)
          Interrupt:5 

eth0.1    Link encap:Ethernet  HWaddr 9C:65:F9:1D:F4:BE  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10792 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:1100964 (1.0 MiB)

eth0.2    Link encap:Ethernet  HWaddr 9C:65:F9:1D:F4:BE  
          inet6 addr: fe80::9e65:f9ff:fe1d:f4be/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1683 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:284213 (277.5 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:6104 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6104 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:659690 (644.2 KiB)  TX bytes:659690 (644.2 KiB)

ra0       Link encap:Ethernet  HWaddr 9C:65:F9:1E:00:7A  
          inet6 addr: fe80::9e65:f9ff:fe1e:7a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:10827035 errors:471 dropped:0 overruns:0 frame:0
          TX packets:1626781 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2346320031 (2.1 GiB)  TX bytes:56836430 (54.2 MiB)
          Interrupt:6 
root@ReSpeaker:/etc/config# brctl show 
bridge name     bridge id               STP enabled     interfaces
br-lan          7fff.9c65f91df4be       no              eth0.1
                                                                           ra0

root@ReSpeaker:/etc/config# route 
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         10.1.1.1        0.0.0.0         UG    0      0        0 apcli0
10.1.1.0        *               255.255.255.0   U     0      0        0 apcli0
10.1.1.1        *               255.255.255.255 UH    0      0        0 apcli0
192.168.100.0   *               255.255.255.0   U     0      0        0 br-lan

and the respeaker to a route which can access internet ,and my phone connect the ReSpeaker's ssid "ReSpeakerXXX",the phone ip is 192.168.100.3 and the phone access 192.168.100.1:3000

xiaooquanwu, 2017-02-24:
The Error URL is 
https://na.account.amazon.com/ap/oa?appActionToken=qlLKAGlKO9yUm3D9cutHTDUdfdYj3D&appAction=actionField&savedParameters=eyJwIjoiS2FGMVJURXY4RW5RZ3Y1d21FVFlTL2VBSDlkbFY3RzhXb3JyUVFrVDFkRHA3ZkdneTZMeFRVZ0ZwOTZJUVR0STBpOE9leTZ0dTNMZC9kejRFalRCaFNTVHpyTDFVVEpNMTUwMWlhQ1R0YzVubG5ZWkFLTDJlQnBZYlduTzYzUWVzUzRvTlY3dGxsSFRwUXBTc0pwOEhabXd6MW1nc05uY3FNRk8vZ2Y4aTlZUjd6NjVSM2NtWXU1bEZWRFZHdDdvbTRmS2tEOFliZVpUazJDSE5PZGZZZkREQWNQalo2OUNJVHdoRERMYVFQUHAvSk9IMDlOYzU4WmZsSks0MzlJNGtmem10ZUcyZytjNVpMaFRUMFo4bkxicUJreEF3RjlzdHlmdG1OSWN1QXh4cFNQSU1iNGhVV011SDJmYzR5UkpaRkoxL3ZmUytYbFhPZXdsK0VmYi9vR2pVVmpHc2dYMDdKSlAzSUJTZDVHamQ4dS9pZ2d5aCtwTk9sc1FZS2Q1NE1YTCtocEhoVGFINGxQZ0lWSnVkU2RxaWFQNUZELzNCL1FvOTJzWkFCREhrUEx6N3A0QllpQUZDeFFKMmNhays2aGpSc0xzNi9FNUN4WVV0cXdmYVFha3Fta3JaRmY5dEhiem8rcHo5SERialFZUTFvR2ZiYVZlVjg5V0RNcE11L0tCQ2FDQlkwU252MFdNVHU2R2dtRU9CaTYyWTBGWkNObUE5RWVYZE14RUNqRTF3eUxKVGZ2OUNPQXI4bC96WDZ5Q2tYU3hMTG9kQ1VRV1VCUlRXdlhWL2h0SUNGR2JuOHd0eFQ1R0RFTjV4QzZVWmdtbHJPYmpKelNocUZjeU0yQXRsT3pmOXlXaXFKYVZzZkd5WXAya1VMN3dacXF4UWFTa0JldE9SbmFESUNGUlVIcCtTMFo3SXlaMGFlb3QxUTVwWEp2UFRIQldyZUdMenpWN29QSGd5YVRqaHRNSHBSdnlVcGY1WThMNXR6ZllWY1YxSlZZOEJYeDdLVUVDald3WFV5WHQrd3VpQWVSOHZnZFlLVWlPS0hYQ1JuZlZtTUVlVEtuMWxqSjgwajk2c1NkZU1hYmZLSFVoS25LZVNSOTN1QXhKeTRLQ08zYlRxU2I5dWFiKzJKUkV2Qnc0SnhXRnNKOXlwSjZZNmFXNExoQ3NvYlVJSkd1SUE4azE0eVJRZG55RjBXaTh5NUdpRk5Qa0phU0czYVpVMStib3lIZ0FZVzFSendoM3VrVlBPYWJqb3M0cGZma1JrRzAxUGZxZWQ3cjRZUWV3T0ZtbXJNOGFRY3c3Yk9EVUw4YzNUUENKMllubFJUTCtCQkVPaStuT0pxc2tnSEF6aStWY2J3WlVGYmw0ZlBnNmR5enhKYzVyeDExKzdEdE5IUWZBOWtSZGhPUG9KRXdKekVpdXNwVmlUNVpQdXFMcGtvdm5CcGJWYlFCZlIrK1Q3bUcxSjRHb0JNZFlDUUdFQStrZ05SRUgzV1NBb2V3aXQ0WVJrVmVxMXJDK1FJZTFmRU9nZzQ4ekRPU084ZVBHeExab0ZPbkhxKzlJQjVDVU1QcEEvOWI2QlRyQTlVRXZUL0srVzNnYW5rTzhjOWNKbnk0dVJYbVFKVFhYN1l3bVQrWm9BRHo4OW9kUU5wZEhSM0JjSVpZd0pSZ2R3cGM5ZFMyV285SGJ4UWZKeEJZZ2pUa29lbDFUT1BCUFBnOU4yWno4Rnp0c081K2hBa2EwYkw5ZFZqdy82T0JiL01rZnpiWi9aR0Z0d3NjdSsrOEIxa01OZ3VkSUxvamwvclhZd3BTaHNPVjI0eWFDRG5sYUtkYmNuMk5WR3kzUGkxMW9IbUlBbW1JeXd5MXNkV0Rxc3ZxdUlLUWNQbk5GZ1BWUlZxVXVIWERJajJkanlmSmFuOXRWU2tMTXNDRU9BcTQ2Vyt0bWZUT1QxNllUUzU4bDFqd3hheFdFcVZ6cGVxbVcxQW1YMVhUUjlCYjV6d3MwM0hORjVVWUM5VVlCSW1uSDI2RnpFNFA1bGJkSTcyamh4V1Y5Ky8wNGxpTlY4L2xJbDFwdGNnb1Y3cWFTT042ejRORnc1ckVIQy8xNEFOZXVHNldUd3IvVEdYQlNSL0M5NWVHL2dEUkZ3SWt0OWpLOGJ6NS9rS1BENXc9PSIsImkiOiJzakJmMDZNQ3EvUjVsdXI3UTkzSndBPT0iLCJzIjoiNDI0RDZBTVBzWjJPa2FCMytPc0ZFSTRHTXFTUExXN0dtUXZpV2pjOXliST0iLCJldiI6MSwiaHYiOjF9&aToken=Atza%7CIwEBIPDt9rFXCZZRLDAf67H-jIdFTyQi_cFr2TtfP5Ebx1OFpsApEd7zAC-cCBywvRFzgeha2mgWp-dwI4kyPZSm4gcOanGoJxMblplKuaCR7qinfIyOdd8Afx0vObkVTjn8XfdzE-PJruLg_rAqkIU6TQyW6Qejh9eV6CJr8atdtGkyT3FZeQw48dvfuEgQ5p0oOpPMiJ81_yHw3Ix_Ca2Xq_Frs_vbsgBpIUDs7G0TnEIoWuV5bGhV8qRzpDy6me5UM-P2VpZU4mddpiMyU7vIupZ1yyeY8fYLVEQ1ssMvzyAxGMC_gp4VrBCb-JHyq-8bolZdu7_m-8tc3yf8WN4-0Tartbgiv_xL1mXom_QgwsYsNgXeYDiPSbtKPpU3H9VYJiaey5HOstBOtyP3bMDnUYYr&openid.pape.auth_policies=http%3A%2F%2Fschemas.openid.net%2Fpape%2Fpolicies%2F2007%2F06%2Fnone&language=zh_CN&consentApproved=

xiaooquanwu, 2017-02-24:
I try 
Allowed Origins: http://10.1.1.287:3000
Allowed Return URLs: http://10.1.1.287:3000/authresponse
is OK ,but the ip is not static ?

xiongyihui, 2017-02-28:
Yes, the IP is not static. It will work if it's on your allowed origins list.

By the way, if you just want to use Alexa, you don't have to setup a alexa developer account, you will be able to use it when you login with your amazon account in via  http://192.168.100.1:3000

The network connection should be:

```
ReSepeaker (two IP: 192.168.100.1 and IP from your router dhcp) <------> Your router <----> Internet
  |
  |
Your Computer or phone
```

Open http://192.168.100.1:3000 in your computer or phone

ishotjr, 2017-07-17:
I was able to get it working as @xiongyihui describes above - at first I tried to use the IP on my own network, which was not whitelisted, but by connecting directly and using http://192.168.100.1:3000/ I was able to get Alexa working as a service with my Amazon consumer account/without having to create a new device on my developer account. 😌

closed 2017-10-17 by KillingJacky