respeaker/get_started_with_respeaker #64: The latest code for the github does not work？

xiaooquanwu, 2017-02-24:
i have update the latest code ，but it can not stream music ，do anyone can fix the bug ？

aleifuu, 2017-02-25:
can you specifically point which code exactly that you updated ?

xiaooquanwu, 2017-02-27:
https://github.com/respeaker/respeaker-feed

xiongyihui, 2017-02-28:
Which application do you use to stream music? Which protocol? AirPlay or DLNA?

xiaooquanwu, 2017-02-28:
DLNA is OK，but play local music through http://10.1.1.218/music/  is not ok ，maybe it is the web page issues ，the web show` Trying to reach {{ programName }} on {{ hostname }}. Please wait...` always

xiongyihui, 2017-02-28:
It seems it's the IP issue. Can you connect to the ReSpeaker's AP and open http://192.168.100.1/music/ ?

xiaooquanwu, 2017-02-28:
the ip 10.1.1.218 is respeaker‘s wan ip ，when connect ReSpeaker directly，and access http://192.168.100.1/music/ ，it is the same issue