respeaker/get_started_with_respeaker #65: Respeaker wakeup error

xiaooquanwu, 2017-02-27:
the Respeaker  is often awakened even no voice

```
INFO:mic:Start detecting
INFO:mic:Detected alexa 
DEBUG:root:wakeup
111111111111111111111111111111100111111101111111111111111111
INFO:mic:Start listening
INFO:requests.packages.urllib3.connectionpool:Starting new HTTPS connection (8): api.amazon.com
WARNING:root:EOF occurred in violation of protocol (_ssl.c:590)
```

aleifuu, 2017-02-27:
same issue I experienced. but only happens when MicArray inserted. Using only the core the sound accuracy is so much better

toxin20, 2017-02-27:
I have the same problem as well. 

When the room is silent and I'm justing using the "alexa" wake up word it is almost never recognized. But if i play some music/video in the room I get false positives all the time

PRONOVATEK, 2017-02-27:
Me too, when I change the word on Alexa example for ReSpeaker I get false positives all the time, or sometimes when I use "Alexa" word too.

alexeinz, 2017-02-28:
i think tuning pocketspinx keyword recognition is needed, devs are not existing but this should be of help
http://cmusphinx.sourceforge.net/wiki/tutoriallm
gonna work on it tonight because otherwise that bloody mic array is useless

aleifuu, 2017-02-28:
@alexeinz , thanks for the URL ... probably what I need to refined my own project further

on the mic array issue, I'm guessing they ( dev/ people ) need to refine further the Microphone class to filter all the noises and what not ... 

dev/ : are you doing it ?

xiongyihui, 2017-02-28:
We can reduce the false positive rate by increase the threshold in the `keyword.txt`, but it also reduce the sensibility.

Another way is to adapt your voice with current acoustic module, more detail via http://cmusphinx.sourceforge.net/wiki/tutorialadapt. 
It will effectively improve individual keyword recognition, but may make the recognizer less general.

xiaooquanwu, 2017-02-28:
midify 1e-30 to  1e-60 is not ok ，it do not fix this issue

xiongyihui, 2017-02-28:
@xiaooquanwu You decreased the threshold.
To reduce false positive rate, we should increase the threshold. For example, modify `1e-30` to `1e-20`.

PRONOVATEK, 2017-02-28:
I try and only respeaker not detect Nothing i modified to 20, 15, and 7 and only with 15 function ok with false positives after a few seconds and of course I modified dic with a tool included on pocketsphinx....

I hope that hoy can help us 

Regards

aleifuu, 2017-03-12:
@xiaooquanwu , does it work yet ?

closed 2017-09-06 by xiongyihui