respeaker/get_started_with_respeaker #66: DEVELOPERS? Are you there?

notnatech, 2017-02-27:
DEVELOPERS? Are you there? No one is looking at any open issues.

aleifuu, 2017-02-27:
+1 on this issue. There have been minimal interaction from the dev/ people on this ReSpeaker. dev/ : hello world ?

alexeinz, 2017-02-27:
+1 too.. there is 0 development or interaction... Alexa is still v1... no hardware issues are addressed ..

aleifuu, 2017-02-27:
hey @notnatech , btw are you still experiencing overheat issue ? 

I try clone the respeaker_hi/ and execute the main.py ... it does spit out a lot of '000000', maybe that's how you get your ReSpeaker overheat unnecessarily ?

notnatech, 2017-02-27:
I turned that debugging off.... I spit out the output to /dev/null so nothing should show up there... But not sure once you call the .listen() command.

xiaooquanwu, 2017-02-27:
DEVELOPERS? Are you there? +1

toxin20, 2017-02-27:
Some communication with us backers/buyers would be nice indeed…

xiongyihui, 2017-02-28:
We are always watching the issues. One reason to keep some answered issues open is that it's not easy to search closed issues.

closed 2017-03-07 by notnatech