respeaker/get_started_with_respeaker #67: juci configuration - disable autologon

alexeinz, 2017-02-28:
Any pointers where have you defined autologon ? 
and how to secure the secondary respeaker network ?

alexeinz, 2017-03-01:
anyone????

m038, 2017-03-01:
What i could find so far is the following:
* Go into Juci
* Click on WIFI (top menu)
* Click on Interface (left menu)
* Click on the interface _Respeaker*****_ (stars are some kind of code)
* Then you can make changes for the Respeaker***** wifi network. 
* Click on apply once you're done

At least add some authentication and perhaps hide it's SSID.

m038, 2017-03-01:
And additionally to protect your web interface:

* Go into Juci
* Click on System (top menu(
* Click on Passwords (left menu)
* At _Change password for user_ in the dropdown select _admin_
* Click the _Change password_ button
* Current password: root
* Enter a proper new password 
* Click _change password_
* Logout 

Now it should ask for a password when trying to access the Juci web interface.

closed 2017-03-05 by alexeinz