respeaker/get_started_with_respeaker #68: How keywords.txt file work

Eternyt, 2017-02-28:
I read many issues about fixing bad word reception, and I suppose that the key of the question are the threshold associated to the words in the keywords.txt file in the respeaker library directory. I search a lot but I continue to not understand how these threshold works. Can someone (/dev?)explain me how they works?

aleifuu, 2017-03-02:
yes, if you could explain just in general what does it do that will be great !

aleifuu, 2017-03-07:
Damn. That was 5 days ago and no one from dev/ give a little bit more explanation ???

xiongyihui, 2017-03-10:
`keywords.txt` contains keywords and their threshold. For example, `keywords.txt` from https://github.com/respeaker/respeaker_python_library/blob/master/respeaker/pocketsphinx-data/keywords.txt is

```
respeaker /1e-30/
alexa /1e-30/
play music /1e-40/
```

`respeaekr` is a keyword, `1e-30` is its threshold. To improve sensitive, we can decrease the threshold, for example, `1e-50`. We should know  decreasing the threshold will increase False Acceptance Rate.

If you want to add new keyword, you should firstly add the keyword to  [`dictionary.txt`](https://github.com/respeaker/respeaker_python_library/blob/master/respeaker/pocketsphinx-data/dictionary.txt).  The `dictionary.txt` is like:

```
respeaker	R IY S P IY K ER
alexa	AH L EH K S AH
play	P L EY
music	M Y UW Z IH K
```

The first part is a name (respeaker, alexa or music), the second part is its phonemes. You can find words in a large dictionary at https://github.com/respeaker/pocketsphinx-data/blob/master/dictionary.txt

Eternyt, 2017-03-10:
Many thanks! But I still don't understand the function of the threshold. Why if I decrease it I improve sensitivity and vice-versa?

xiongyihui, 2017-03-10:
You don't have to. If you yell `respeaker` several times but the pocketsphinx doesn't recognize it, you should consider to descrease the threshold.

Eternyt, 2017-03-10:
And how much I must decrease/increase the threshold? Randomly?

aleifuu, 2017-03-12:
I'm using the `keyword.txt` by default when with the ReSpeaker core. and it works fine recognizing it from a meter away 

However, as raised several post before, when attaching MicArray with it, it seems ( in layman terms ) becoming stupid in recognizing keyword from a distance ( let's say 3 or 4m away )

How much should I decrease/increase the threshold in this case ? I really want ReSpeaker to be able to work and recognize command/wake up keyword at least anywhere if placed in a small/medium room

snizzleorg, 2017-04-16:
Where do I find the keywords.txt file on the filesystem?

jerryyip, 2017-04-17:
@snizzleorg 
`vi /usr/lib/python2.7/site-packages/respeaker-0.6.0-py2.7.egg/respeaker/pocketsphinx-data/keywords.txt`

sudhaa1232, 2017-04-21:
myine is little different. use following command

find . -name keywords.txt

closed 2017-10-17 by KillingJacky