respeaker/get_started_with_respeaker #69: power requirements including mic array

alexeinz, 2017-02-28:
Hi guys, I wonder what are the power requirements
I know that with the mic array , the current apple 1A 5V adapters are not pulling enough juice
have u done any testing or have recommended adapters?

jerryyip, 2017-03-01:
Hi,

Only mic array->300mA
mic array with led on->500mA
respeaker core with mic array and len on->Max 1A

Did your respeaker always need more than 1A when the mic array is connected on?

alexeinz, 2017-03-01:
hmm it seems not enough for me for some reason.
I need to do some testing on how much my macpro usb port puts out but 1A apple doesnt push enough juice to start it up with the mic array... currently using one of em larger ipad adapters.

jerryyip, 2017-03-01:
The apple support said that USB 3 provides up to 900mA at 5V.
https://support.apple.com/en-us/HT204377

alexeinz, 2017-03-01:
not really.. it can be more too if you read below, anyways i just need to test how much its drawing because the difference is too strange.

alexeinz, 2017-03-01:
yea 100% not enough on normal power

aleifuu, 2017-03-02:
Hey there, I'm glad someone raised issue about power requirement because certainly there was none mentioned in the README.md section. 1A maximum yes ? Will there be a problem if your charger spits out 2A instead of 1A ?

jerryyip, 2017-03-10:
@alexeinz 2A is the rated current of your charger and the actual output current depends on the load(max 1A).

closed 2017-04-16 by jerryyip

RakshithNP, 2017-11-22:
@aleifuu, you can use even 100A charger no problem, as the load only consumes the required power.
but the voltage should not be above the safe level (i.e it should be around 5 V only) or else you may fry up your respeaker (load)