respeaker/get_started_with_respeaker #70: ReSpeaker keep red led on after boot.

Eternyt, 2017-03-01:
I use a while offline thinking that the LEDs signals the absence of the internet connection, and now also with wifi when it boot up the red LEDs don't go off...

jerryyip, 2017-03-02:
The LEDs are controlled by the arduino 32u4. The reason they don't go off is that openwrt has not send a "turn led off" command via spi to arduino.  You could turn the leds off by yourself with [Python API](https://github.com/respeaker/get_started_with_respeaker/blob/master/ReSpeakerPythonAPI.md#class-pixelring).

Eternyt, 2017-03-02:
Yes I know, but I want to understand why before the leds are turned off and now not.

jerryyip, 2017-03-10:
LEDs still don't go off when it boot up?

Eternyt, 2017-03-10:
No, now LEDs go off. But I don't know why...

closed 2017-03-10 by Eternyt