respeaker/get_started_with_respeaker #71: Memory error during Speech recognition module for Python instalation

ppapagay, 2017-03-02:
Hi,

I tried to install Speech recognition module for Python, but I got this Memory error. Can you help me please?

root@ReSpeaker:~# pip install https://github.com/xiongyihui/speech_recognition/a
rchive/respeaker-lite.zip
Downloading/unpacking https://github.com/xiongyihui/speech_recognition/archive/respeaker-lite.zip
  Downloading respeaker-lite.zip (unknown size): 39.4MB downloaded
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/pip-1.5.6-py2.7.egg/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/site-packages/pip-1.5.6-py2.7.egg/pip/commands/install.py", line 278, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python2.7/site-packages/pip-1.5.6-py2.7.egg/pip/req.py", line 1197, in prepare_files
    do_download,
  File "/usr/lib/python2.7/site-packages/pip-1.5.6-py2.7.egg/pip/req.py", line 1375, in unpack_url
    self.session,
  File "/usr/lib/python2.7/site-packages/pip-1.5.6-py2.7.egg/pip/download.py", line 582, in unpack_http_url
    unpack_file(temp_location, location, content_type, link)
  File "/usr/lib/python2.7/site-packages/pip-1.5.6-py2.7.egg/pip/util.py", line 621, in unpack_file
    unzip_file(filename, location, flatten=not filename.endswith(('.pybundle', '.whl')))
  File "/usr/lib/python2.7/site-packages/pip-1.5.6-py2.7.egg/pip/util.py", line 495, in unzip_file
    data = zip.read(name)
  File "/usr/lib/python2.7/zipfile.py", line 935, in read
    return self.open(name, "r", pwd).read()
  File "/usr/lib/python2.7/zipfile.py", line 630, in read
    data = self.read1(n)
  File "/usr/lib/python2.7/zipfile.py", line 665, in read1
    data = self._fileobj.read(nbytes)
MemoryError

Storing debug log for failure in /root/.pip/pip.log

xiongyihui, 2017-03-03:
Did you insert the sd card and use it as extroot (https://github.com/respeaker/get_started_with_respeaker/blob/master/QuickStart.md#use-sd-card-to-extend-storage)?

closed 2017-03-10 by jerryyip

svanharmelen, 2017-07-15:
@xiongyihui yes I did, but I get the same error:

[image]

xiongyihui, 2017-07-17:
pocketsphinx is already installed. If you want to re-install it, you need to cross-compile it.

svanharmelen, 2017-07-17:
Yes, I noticed... Thanks.

About cross compiling, where or how can I get a mips cross compiler toolchain for macOS?

xiongyihui, 2017-07-18:
The toolchain can be downloaded at https://docs.labs.mediatek.com/resource/linkit-smart-7688/en/downloads

m038, 2017-10-17:
Could you provide either a guide how to cross-compile this module for the Respeaker or perhaps provide a compiled copy of that module which can be manually installed on the Respeaker?

With the help of this article http://whatschrisdoing.com/blog/2009/10/16/cross-compiling-python-extensions/ i've come up with the following script.
https://github.com/m038/pocketsphinx-python/blob/cross-compile/build.sh
With all the correct software en pip modules install it works till i get this error:

```
In file included from /usr/include/python2.7/Python.h:8:0,
                 from sphinxbase/swig/sphinxbase_wrap.c:11:
/usr/include/python2.7/pyconfig.h:83:5: error: #error unknown multiarch location for pyconfig.h
 #   error unknown multiarch location for pyconfig.h
     ^
In file included from /usr/include/python2.7/pyport.h:4:0,
                 from /usr/include/python2.7/Python.h:58,
                 from sphinxbase/swig/sphinxbase_wrap.c:11:
/usr/include/python2.7/pyconfig.h:83:5: error: #error unknown multiarch location for pyconfig.h
 #   error unknown multiarch location for pyconfig.h
     ^
In file included from /usr/include/python2.7/pymath.h:4:0,
                 from /usr/include/python2.7/Python.h:77,
                 from sphinxbase/swig/sphinxbase_wrap.c:11:
/usr/include/python2.7/pyconfig.h:83:5: error: #error unknown multiarch location for pyconfig.h
 #   error unknown multiarch location for pyconfig.h
     ^
sphinxbase/swig/sphinxbase_wrap.c: In function 'init_sphinxbase':
sphinxbase/swig/sphinxbase_wrap.c:7671:3: warning: implicit declaration of function 'PyEval_InitThreads' [-Wimplicit-function-declaration]
   SWIG_PYTHON_INITIALIZE_THREADS;
   ^
error: command '/home/mischa/toolchain/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/mipsel-openwrt-linux/bin/gcc' failed with exit status 1
```
I could not find a _python.h_ file in the toolchain provided by mediatek. And i'd have no idea where to get one for this platform.

Yumanook, 2018-08-12:
try "pip --no-cache-dir install SpeechRecognition"

firmanhidayat, 2020-05-06:
> try "pip --no-cache-dir install SpeechRecognition"

it's working well